csete/gpredict #179: VFO mode

mdblack98, 2019-10-14:
GPredict should add an option for the VFO mode for rigctld.  It avoids all the VFO swapping that can occur.  All you do is add the VFO to the remote commands.

So instead of
V VFOA
F 14070000
V VFOB
F 14071000
You do
F VFOA 14070000
F VFOB 14071000

de Mike W9MDB

mdblack98, 2019-10-17:
And I'll add that you can check if vfo mode is enabled
\chk_vfo
CHKVFO 1

If chk_vfo returns an error then it's an old rigctld that doesn't support chk_vfo but may still be in VFO mode.

de Mike W9MDB

csete, 2019-10-26:
Hi @mdblack98 

Thanks for the request. Can you provide some more info about which radio this is relevant for? My concern is that every time we fix or improve something like this for one radio, it breaks the functionality for another radio, and we have been going back and forth for years :(

mdblack98, 2019-10-26:
Here's what I got on another email list from va2cst@outlook.com -- he's seeing VFO swapping which probably happens on a lot of rigs.The vfo mode of rigctld avoids the swapping for any rigs that have director VFO addressing which is pretty much all new rigs now and most of the old ones.

In this mode, both up/downlink are being corrected but the VFOs are Always swaping from main to sub and the SPLIT function turn on.

 

In this mode, both up/downlink are being corrected but the VFOs are Always swaping from main to sub and the SPLIT function turn on.
 

In this mode, both up/downlink are being corrected but the VFOs are Always swaping from main to sub and the SPLIT function turn on.
 

In this mode, both up/downlink are being corrected but the VFOs are Always swaping from main to sub and the SPLIT function turn on.

Aleziss, 2019-10-28:
Hell everyone, I see that someone posted an error that I have reported so thank you for taking care of this.

You can see the actual bug along with some debug capture text from hamlib.

See my bug report here and tell me if I can provide more information on what is happeneing.

https://github.com/csete/gpredict/issues/181

Thank you all !