csete/gpredict #180: Rotator Control:  Antenna sometimes defaults to South

royldean, 2019-10-27:
On about 50% of passes, the antenna controller defaults to 0deg elevation, 180deg azimuth until the satellite appears over the horizon.   This is not a huge deal as you only lose maybe 30 seconds of satellite pass time while the antennas rotate to the proper azimuth, but seems to be new since upgrading to 2.3.57-2d80.

On the rest of the passes, the antenna will correctly park at 0deg (or 180deg) elevation at the proper azimuth to be ready to start Rx'ing the satellite immediately as it rises over the horizon.

csete, 2019-10-27:
Hi @royldean 

Thanks for the report. Do you remember which commit you upgraded from?

royldean, 2019-10-28:
Last installed version was 2.2.1.    I have to admit, however, that I also re-installed ubuntu 18.04 prior to installing 2.3.57-2d80.   The laptop that it is installed on is "flaky", and I suppose this could be something wacky happening on the laptop (and/or hamlib).

csete, 2019-10-30:
Okay, thanks for the info. I will take a look next time I take a "Gpredict sprint" hopefully very soon :)

royldean, 2020-07-30:
Just an update for Alex, as this may have been a local install issue.   I have been using the same version (2.3.57-2d80) since I posted the original problem, at a minumum of once a week to as much as 7 days a week (in other words, constant use).   Yesterday I noticed that GPredict is NOT showing this problem anymore.   I've worked about 10 satellite passes that would normally have shown this issue, yet the tracking is working flawlessly now.

I only changed ONE thing (on Tuesday night).   I uninstalled FLDIGI, then reinstalled it, along with installing FLMSG.   Of course, I do apt udate and apt upgrades all the time, so that might have been the kicker, too.   In any case, this might be a closed issue....  I'll report back in a few weeks if nothing has changed.

royldean, 2020-07-30:
Annnnnddddd of course the very next pass it shows up.   Ignore previous post.