csete/gpredict #182: feature request: push radio mode and tone

Aleziss, 2019-10-28:
Hello all,

That request is all about the transponder data.

1) Have the ability to manually update or edit transponder data of satellites

2) Have the ability to set mode (FM/SSB/CW) for each uplink and downlink

[image]

3) Have gpredict push the radio VFO mode and tone when activating doppler control.

I was also wondering, where does the transponder data come from ? I see no configurable link about the transponder data unlike the TLE elements where we can set links to get the data.

Thank you !

csete, 2019-10-30:
Thanks for the suggestion.

Currently, transponder data is fetched from https://db.satnogs.org/ where anybody can submit suggestions for changes. You will need to create an account though.

Will also take a look at setting mode, etc; however, as these are convenience feature (correct me if I'm wrong) it might not get the highest priority as long as there are critical issues with the software..

Aleziss, 2019-10-31:
@csete Thank you for the reply. I will take a look at satnogs see how it is working.

Yes, pushing operating mode and tone is a convenience feature so I understand that it is less important than at least make the doppler control work properly on the 910, 9100 and 9700,

I've seen the frequency and tone being pushed by satpc32 (another software I did try but hated) and macdoppler (a friend is running this on his 9700) and it is worderful with macdoppler. 

With macdppler, you need to leave the radio out of satellite mode, leave both VFO as A/B (no into memory mode) and macdoppler take care of the rest.

Correct frequency of both VFOs, push mode and tone, even shut off the sub VFO when you chose the beacon frequency.

By the way, gpredict seem to disconnect from hamlib when you chose the beacon frequency as both VFOs are on the same band which cannot happend in the 9700 so I guess it create a miss reading of the data and gpredict closes the connection.

So at this very moment, I've noticed that if the 9700 is NOT in satellite mode, the uplink/downlink are on the wrong VFOs, in order to make the VFOs right, I would need to switch the 9700 into satellite mode, there uplink/downlink will be on the right VFOs.

I thought I could swap those with this parameter but it does not seem to affect anything on the 9700.

[image]

In normal mode, TX is top VFO and RX is bottom VFO. In satellite mode, it is the opposite, the top VFO is downlink and bottom VFO is uplink. You can SWAP UHF/VHF frequencies on the VFOs but TX/RX always stays on the same VFO depending on which normal/satellite mode you chose.

Where I am getting at with all this is that if this feature is not implemented for the moment but the doppler would be fixed, I would program satellite memories with the right modes and tones and let gpredict only read/write the frequency until my suggestion is implemented.

Aleziss, 2019-11-16:
@csete I submitted the idea on satnog for dedicated MODES for each up/downlink [HERE](https://gitlab.com/librespacefoundation/satnogs/satnogs-db/merge_requests/435) and it seem to please users. Follow up can be found [HERE](https://community.libre.space/t/uplink-downlink-modes/4916/9).

For gpredict, this mean it would give the ability to, in a near future hopefully, push mode and tone to a radio ! Not sure what will happend when this is implanted in satnog for gpredict though until this is updated in gpredict.

It seem to be moving forward !

csete, 2019-11-16:
Thanks @Aleziss 

I really appreciate that you put effort into it. It shows that you are motivated to have this feature added, which at the end will have influence when prioritizing things to implement.

Aleziss, 2019-11-29:
It seem to be moving pretty nicely on satnog, I think the option for clear up/downlink modes will be implemented soon. Programming seem to be done, just need some approval and roll out. Great news for the future of gpredict that will be able to use all that data to control radio frequencies, modes and tone.

Aleziss, 2019-12-10:
deckbsd replied this on december 3rd 2019 👍 

[quote="deckbsd, post:12, topic:4916"]
@aleziss The functionality code has been merged. This feature will be available at the next release of satnogs-db :slight_smile:
[/quote]

So soon enough, there will be clear uplink and downlink modes describe in satnog transponder data and hope to see this implanted also in gpredict sometime down the road !

whattwood, 2021-04-17:
As an everyday gpredict user this would be my TOP feature request :-) I can't tell you how many times I've been confused as to why I can't find myself on a sat only to realize I'm in USB instead of LSB, or my CTCSS tone is missing for SO-50 etc.

Thx for everything!