csete/gpredict #183: Engange radio and rotator controller when TCP connection is re-established

csete, 2019-11-22:
https://community.libre.space/t/automatic-engage-radio-device-near-aos-time/4998

jmason86, 2019-11-22:
Or alternatively, maybe a programmatic method for triggering the radio "engage". For example, at the shell command line: 
```
gpredict radio_track
gpredict radio_engage
```

jmason86, 2019-12-02:
I've got a hack working with [PyAutoGui](https://pyautogui.readthedocs.io/en/latest/introduction.html#examples). Just needed a screenshot of the button (Engage in this case). 

Another alternative feature request would be to include keyboard shortcuts for the Track and Engage buttons. Could be the same shortcut for Radio and Rotor controllers and apply to whichever is in focus. That would enable the use of [xdotool](http://xmodulo.com/simulate-key-press-mouse-movement-linux.html#comments).