csete/gpredict #184: Transponder Data

pr0xibus, 2019-11-23:
Morning folks

I have downloaded the AppImage from Git, This was previously working fine in my old Slackware Box, but decided to run Fedora 31 on a Laptop.

Anyway Pulling the TLE from network works fine, but transponder data doesnt seem to pull anything.

Looking at the logs the error is

`2019/11/23 08:22:59|1|modes_update_from_network: Error fetching https://db.satnogs.org/api/modes/?format=json (Problem with the SSL CA cert (path? access rights?))
2019/11/23 08:22:59|1|modes_update_from_network: Could not fetch frequency files from network
2019/11/23 08:22:59|1|trsp_update_from_network: Error fetching https://db.satnogs.org/api/transmitters/?format=json (Problem with the SSL CA cert (path? access rights?))
2019/11/23 08:22:59|1|trsp_update_from_network: Could not fetch frequency files from network`

Accessing the site via Firefox seems to work fine, I thought populating it via

`curl -X GET "https://db.satnogs.org/api/transmitters/?format=json" > transmitters.json`

would do the job but no transponder data is displayed on any satellite.

Any ideas?

csete, 2019-12-02:
Hi,

Sorry for the late reply, but I'm not monitoring this issue tracker frequently since we only use it for development and not user support.

I suspect the issue can be some SSL library or certificates missing, perhaps libcurl4-openssl or similar. Perhaps somebody who monitors the forum or the mailing list is more familiar with fedora: http://gpredict.oz9aec.net/User_Support.php

barisdinc, 2020-01-30:
I can take care of this.... 
Let me reproduce the case first....

barisdinc, 2020-02-13:
@pr0xibus hi,
I am trying to reproduce the problem in my local but I cannot.
Could you please share more information about your environment.

Thanks
Baris
TA7W /OH2UDS

closed 2020-10-28 by csete