csete/gpredict #185: Feature Request: Bulk selection of satellites

mark-orion, 2019-11-30:
Please add functionality that allows the selection of multiple satellites at once to the "New Module" dialog. Having to add each satellite one by one is very laborious.
I would like to see the use of the standard bulk editing options:
Ctrl + Click adds satellite to selection
Ctrl + A adds all satellites in the list on the left hand side
Mark first satellite and then Shift + Click selects a range of satellites

csete, 2019-12-02:
Hi,

These shortcuts were readily available but were disabled a long time ago because the application was not very stable with many satellites in one module. However, it is more than ten years ago, so perhqaps it is better now as computers have become faster and have more RAM.

Wil ltake a look at it at some point.

zberry, 2020-07-21:
I would also really like this feature.

minecraft2048, 2023-02-02:
We want to add all Starlink satellites into Gpredict but now we need to click the add button like several thousand times. Is there a workaround somewhere where we can add all the Starlink?

mdblack98, 2023-02-02:
Can't you just add https://celestrak.org/NORAD/elements/supplemental/starlink.txt to the TLE list?

Mike W9MDB

mdblack98, 2023-02-02:
Just realized you mean in the Edit module for the satellites.

Are you compiling gpredict yourself?

I should be able to make that field a multi-select field on my fork of gpredict.

Mike W9MDB

minecraft2048, 2023-02-02:
That is what I want, getting multi select / shift select on all the Starlink sats 
Another useful thing is a button to show/hide all ground tracks and footprints