csete/gpredict #186: Please update Ubuntu PPA with latest version 2.3.x builds

mark-orion, 2019-12-01:
I notice that the Ubuntu PPA is still on version 2.2.1. It would be great to be able to run the latest GPredict versions on Ubuntu 18.04.

csete, 2019-12-02:
The latest officially released version is 2.2.1. Version 2.3 refers to the development series and is not an official release tag. As much as I love the concept of "release often" I do not have the time to test, package, and support frequent development snapshots at this time. I can do that later when I will focus on gpredict, but right now I have too much work and other projects to work on.

mark-orion, 2019-12-03:
>  I can do that later when I will focus on gpredict, but right now I have too much work and other projects to work on.

You have my sympathy - similar situation here. Thanks for all the work !

csete, 2020-10-28:
The PPA will be updated as part of the release process when we have a release.

closed 2020-10-28 by csete