csete/gpredict #191: Setting a Default transponder in Radio Control? it always defaults to First transponder in the drop-down list

dxblouie, 2020-02-12:
Hi all,

so i'm not sure if this is a bug, a feature, or simply not something addressed..
I'm tracking NOAA 15, 18 and 19.. using prediction and auto tracking to tune SDR# and handle doppler shift for the 3 satellites when in range.

i record APT signal to get low-res weather imagery, which works perfectly with NOAA18 and 19, both of which have APT Transponder on top of the list.

[image]

NOAA15 however has "ARGOS-3 DCS" on top, and whenever NOAA15 is in range, radio control tunes to ARGOS-3 transponder..

[image]

the issue is simple, if i'm not there to change the transponder to APT, gpredict tunes my receiver to the wrong transponder, and i miss the pass..

am i missing something? a setting so the selected transponder would be the default for the satellite?

Thanks in advance!

srcejon, 2020-02-13:
I don't think there is a way in the GUI, but you can change the order by editing the corresponding trsp file in a text editor. For NOAA 18, look for trsp/28654.trsp

dxblouie, 2020-02-13:
i did contemplate that idea, but figured an automatic update would take me back to square 1 :)

i edited 25338.trsp (for NOAA 15) to move this transponder to the top.. it works, till the next update!

[APT]
DOWN_LOW=137620000
MODE=APT

csete, 2020-03-22:
I think it will be useful to save which transponder was used last time and select that one by default the next time the radio controller is opened. Will try to have this done in the next release.