csete/gpredict #193: LOS isn't signalled if autotrack is enabled

srcejon, 2020-02-18:
It seems LOS isn't signalled if autotrack is enabled (or at least sometimes isn't signalled). 

This appears to be because update_autotrack calls gtk_rig_ctrl_select_sat, which doesn't call check_aos_los before changing the satellite. 

Thus if update_autotrack detects the satellite LOS before check_aos_los does, LOS will not be signalled, as the next time check_aos_los is called, the satellite will have already been changed to the new satellite.

srcejon, 2020-02-25:
Possible fix included in https://github.com/csete/gpredict/pull/192/commits/d391ce92ff0e3120a5273e3c70b047e4f7b33d45

Call check_aos_los() in gtk_rig_ctrl_select_sat()

void gtk_rig_ctrl_select_sat(GtkRigCtrl * ctrl, gint catnum)
{
    sat_t          *sat;
    int             i, n;

+    /* This is called indirectly by update_autotrack, when a satellite goes
+       below the horizon. So we need to call check_aos_los here, to give it a
+       chance to signal LOS, otherwise it will be missed, as the current sat
+       will have changed on the next call. */
+    check_aos_los(ctrl);

Valstee, 2021-03-25:
Hi, I've got the same problem.
Did you try your solution ?
Thanks,
Valstee