csete/gpredict #199: Gpredict Module Configuration error

Eric731-art, 2020-02-25:
Hello from Eric/KB3DTQ 
I'm working with the SARCTRAC Rotor controller and actually was successful in building and running Gpredict program/app on my new Dell Inspiron Chromebook model 3181📡👩🏻‍🚀🛰.
This how my setup looks:
1. I'm using strictly Linux (Beta) in the Chrome OS (not Windows 10)
2. Arduino Pro Micro ATmega32U4 5volt 16MHz board
3. DSD TECH SH-U09C2 USB to TTL Adapter as a connection to Rotor 
Problem: in Gpredict when I try to Clone, Configure Module I'm getting an Error message.
All data fields: Name, Description, Location, Longitude, Latitude, etc are correct, but will not Configure and Save. Any guides or solutions would be great appreciated.

csete, 2020-02-26:
Hi Eric,

Perhaps the application has no rights to write to the disk. It is hard to say without knowing what the error message is.

In any case, please try using the user support forum or the mailing list, see http://gpredict.oz9aec.net/User_Support.php
We only use github for development and not for user support

Eric731-art, 2020-02-27:
Ok thanks so much for the follow up Alex. Her is a screen shot of Gpredict
Module error message. I'm sure I'll figure it out some way. Plus I have a
lot of CMU students that live around me.
Many Regards, Erich

csete, 2020-10-28:
Closing it as I don't think this is a problem in gpredict. Feel free to follow up if the persists.

closed 2020-10-28 by csete