csete/gpredict #201: Segmentation error on Raspberry Pi Zero WH with Raspbian GNU/Linux 9 (stretch)

rutja76, 2020-03-06:
When trying to install gpredict_2.2.1-ubuntu18.04_ppa3_armhf.deb I get a segmentation error. Debugging with gbd I get:

```
Starting program: /usr/bin/gpredict 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x0045a1c2 in ?? ()
(gdb) bt
#0  0x0045a1c2 in ?? ()
#1  0x0040b00e in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

However, Gpredict 1.3 works flawlessly, problem seems to shop up with GPredict 2.2.1. I get the same error also with gpredict_2.2.1-ubuntu16.04~ppa2_armhf.deb

Eric731-art, 2020-03-06:
This is just update info regarding Gpredict running on Dell Linux/Beta
Chromebook. When I downloaded/installed Ubuntu 14 the whole Gpredict App
was refreshed and looks great. So far everything runs normal, Modules look
normal and the actual App name shows up as Gnome. I'm still tweaking the
rotor engage part. But it looks great, I installed/built the App on Linux
Beta terminal.📡👩🏻‍🚀🛰
73's Eric/KB3DTQ

rutja76, 2020-03-07:
UPDATE: I get the same segmentation fault also after upgrading to Raspbian GNU/Linux 10 (buster)

rutja76, 2020-03-07:
UPDATE 2

After upgrading to Buster, running `sudo apt-get install gpredict` will install directly Gpredict 2.3-33-gca42d22-1. And in works. The problem above arose if trying to install gpredict_2.2.1-ubuntu18.04_ppa3_armhf.deb downloaded from the repository. I'm closing this issue as `sudo apt-get install gpredict` installs successfully the latest version after upgrading to Buster.

closed 2020-03-07 by rutja76