csete/gpredict #211: [Feature][Question] GPredict server

PVaub, 2020-04-19:
Hi,
I was wondering if you ever though about separate the back end an the front end of GPredict ?
Indeed, by creating a server or a command line interface, it will be easier for some users (like me) to integrate it into some projects.
So have you ever though about it and if yes was there any obstacle (except the time it would take) ?

Thank you in advance.

PVaub, 2020-04-19:
Oh and by the way I've seen that GPredict was build on Predict (https://www.qsl.net/kd2bd/predict.html) which have a CLI but I'm afraid of its potential instability and a lack of functionalities and optimizations for modern computers.
Maybe there are some other open softwares that run on Linux with CLI/server mode but I haven't seen any of them during my searches.

csete, 2020-04-19:
This is not currently on the roadmap, sorry.

If you wish to have discussions or need technical support, please use the forums dedicated to that: http://gpredict.oz9aec.net/User_Support.php

closed 2020-04-19 by csete