csete/gpredict #214: Gpredict doesn't support gpsd in Windows 10.

iehx, 2020-04-20:
Gpredict doesn't support gpsd in Windows 10.
I'm going to DIY an antenna rotator, have bought an  axis gyroscope with GPS.
Could the developer add the gpsd module to gpsd in win10, thanks a lot.

csete, 2020-04-20:
Sorry, but I'm not sure I understand what you mean. Do you mean to include gpsd support in the distributed windows binaries?

Gpsd support is still under development and not included by default, see issue #2. Also note that gpredict only uses geogrphical location data from gpsd and will not use any gyro data if that is what you are after. So it is only useful if you have a portable ground station.

iehx, 2020-04-21:
Thanks for reply.
Yes,I mean include gpsd support in the distributed windows binaries.
I'm a newcomer to using github and compile the code.
What's the tools for complie the program in windows 10.
Thanks again.

iehx, 2020-04-22:
> Sorry, but I'm not sure I understand what you mean. Do you mean to include gpsd support in the distributed windows binaries?
> 
> Gpsd support is still under development and not included by default, see issue #2. Also note that gpredict only uses geographical location data from gpsd and will not use any gyro data if that is what you are after. So it is only useful if you have a portable ground station.

help.I'm traped in compile the gpredict in msys2.
 ./autogen.sh runs ok.
make runs fail
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... configure: error: in `/doc/GitHub/gpredict':
configure: error: cannot run C compiled programs.
If you meant to cross compile, use `--host'.
See `config.log' for more details
make: *** [Makefile:402：config.status] 错误 1

config.log
.................
configure:3443: checking whether we are cross compiling
configure:3451: gcc -o conftest.exe    conftest.c  >&5
In file included from conftest.c:11:
C:/msys64/mingw64/include/stdio.h:5:11: fatal error: libio/stdio.h: No such file or directory
    5 | # include <libio/stdio.h>
      |           ^~~~~~~~~~~~~~~
compilation terminated.
configure:3455: $? = 1
configure:3462: ./conftest.exe
./configure: line 3464: ./conftest.exe: No such file or directory
configure:3466: $? = 127
configure:3473: error: in `/doc/GitHub/gpredict':
configure:3475: error: cannot run C compiled programs.
If you meant to cross compile, use `--host'.
See `config.log' for more details

## ---------------- ##
## Cache variables. ##
## ---------------- ##
................................................................
thanks a lot.

iehx, 2020-04-25:
I'v got it.
1st use msys2(default) or msys32,not msys64,update the newest version includ gtk+.
2nd cd predict/win32
3rd make
After a while,the binaries will be created.

EeeLo, 2022-03-12:
I was so tird with the MSYS2 ,there have 32 version and 64 version, the doucument not say wich one we need and the link about gtk+ is down.

EeeLo, 2022-03-12:
# make
i686-w64-mingw32-windres --define __WIN32__ --define __WIN95__ --define MSRC --define __GNUWIN32__ --define VERMAJ= --define VERMIN= --define VERBLD= gpredict.rc gpredict_res.o
D:\msys64\mingw32\bin\i686-w64-mingw32-windres.exe: gpredict.rc:25: syntax error

wkitty42, 2022-03-12:
why are you defining VERMAJ, VERMIN, and VERBLD and leaving them empty?
perhaps this will work better since the rc file defines them properly if they do not already exist...
`i686-w64-mingw32-windres --define WIN32 --define WIN95 --define MSRC --define GNUWIN32 gpredict.rc gpredict_res.o`