csete/gpredict #227: Gpredict disengages when changing RX freq manually while transmitting (using two rigs)

rutja76, 2020-07-09:
I’m testing a configuration with one rig for TX (FT-817) and one for RX (TS-2000). The idea is to have possibly a HT for RX (like TH-D74) but for now TS-2000 is all I have.

It’s known that FT-817 doesn’t update the frequency while transmitting. However, having another radio for the RX, I can change the RX freq manually in order to compensate the drift caused by the fixed TX freq.

However, after changing the RX freq manually through the knob of the TS-2000 for a couple of seconds while transmitting with the FT-817, Gpredict disengages by itself.

I can confirm that this doesn’t occur in RX situation, so to summarize:

- RX: Tuning the RX radio knob (TS-2000) everything works fine

- RX: Turning the TX radio knob (FT-817) everything works fine

- TX: Turning the TX radio knob (FT-817) everything works fine

- TX: Tuning the RX radio knob (TS-2000) Gpredict disengage if tuning lasts for a couple of seconds. If I just quickly turn the knob it remains engaged.

- TX: normal transmission without manual tuning works fine

Here belowis the debug of the last commands of rigctld before the disengage. I have put the two logs (TS-2000 for RX, FT-817 for TX) in one since the commands appear to happen consecutively without overlaps. The cycle to 3000 ms in order to have a shorter log. It appears that FT-817 (TX) disengages first and then TS-2000 (RX) follows.

Does anybody see anything that could give a hint on why Gpredict disengages?

_[TS-2000 (RX rig)]_

2020-07-09:06:11:16.352029: client lock engaged
2020-07-09:06:11:16.352158: rig_strvfo called
2020-07-09:06:11:16.352212: rigctl(d): f 'currVFO' '' '' ''
2020-07-09:06:11:16.352265: rig_get_freq called
2020-07-09:06:11:16.352314: kenwood_get_freq called
2020-07-09:06:11:16.352375: kenwood_safe_transaction called
2020-07-09:06:11:16.352431: kenwood_transaction called
2020-07-09:06:11:16.352484: kenwood_transaction: cmdstr = FA
2020-07-09:06:11:16.352538: serial_flush called
2020-07-09:06:11:16.352604: write_block called
2020-07-09:06:11:16.403201: write_block(): TX 3 bytes
2020-07-09:06:11:16.403359: 0000    46 41 3b                                            FA;             
2020-07-09:06:11:16.403422: read_string called
2020-07-09:06:11:16.465922: read_string(): RX 14 characters
2020-07-09:06:11:16.466079: 0000    46 41 30 30 31 34 35 39 39 32 37 36 31 3b           FA00145992761;  
2020-07-09:06:11:16.466196: client lock disengaged

_[FT-817 (TX rig)]_

2020-07-09:06:11:16.569567: client lock engaged
2020-07-09:06:11:16.569692: rig_strvfo called
2020-07-09:06:11:16.569744: rigctl(d): F 'currVFO' '435343897' '' ''
2020-07-09:06:11:16.569827: rig_set_freq called
2020-07-09:06:11:16.569883: ft817: requested freq = 435343897.000000 Hz
2020-07-09:06:11:16.569963: to_bcd_be called
2020-07-09:06:11:16.570019: rig_force_cache_timeout called
2020-07-09:06:11:16.570068: write_block called
2020-07-09:06:11:16.570515: write_block(): TX 5 bytes
2020-07-09:06:11:16.570634: 0000    43 53 43 90 01                                      CSC..           
2020-07-09:06:11:16.570690: read_block called
2020-07-09:06:11:16.638274: read_block(): RX 1 bytes
2020-07-09:06:11:16.638409: 0000    f0                                                  .               
2020-07-09:06:11:16.638465: ft817: ack received (240)
2020-07-09:06:11:16.638523: client lock disengaged

_[TS-2000 (RX rig)]_

2020-07-09:06:11:19.527218: client lock engaged
2020-07-09:06:11:19.527347: rig_strvfo called
2020-07-09:06:11:19.527400: rigctl(d): f 'currVFO' '' '' ''
2020-07-09:06:11:19.527456: rig_get_freq called
2020-07-09:06:11:19.527506: kenwood_get_freq called
2020-07-09:06:11:19.527561: kenwood_safe_transaction called
2020-07-09:06:11:19.527612: kenwood_transaction called
2020-07-09:06:11:19.527663: kenwood_transaction: cmdstr = FA
2020-07-09:06:11:19.527717: serial_flush called
[... 108 lines cut ...]

_[TS-2000 (RX rig)]_

2020-07-09:06:11:25.944486: rig_open called
2020-07-09:06:11:25.944621: client lock engaged
2020-07-09:06:11:25.944673: rig_close called
2020-07-09:06:11:25.944720: kenwood_close called
2020-07-09:06:11:25.944769: kenwood_set_trn called
2020-07-09:06:11:25.944815: kenwood_transaction called
2020-07-09:06:11:25.944865: kenwood_transaction: cmdstr = AI0
2020-07-09:06:11:25.944919: serial_flush called
2020-07-09:06:11:25.944985: write_block called
2020-07-09:06:11:25.995695: write_block(): TX 4 bytes
2020-07-09:06:11:25.995855: 0000    41 49 30 3b                                         AI0;            
2020-07-09:06:11:25.995918: write_block called
2020-07-09:06:11:26.046396: write_block(): TX 3 bytes
2020-07-09:06:11:26.046570: 0000    49 44 3b                                            ID;             
2020-07-09:06:11:26.046631: read_string called
2020-07-09:06:11:26.070757: read_string(): RX 6 characters
2020-07-09:06:11:26.070900: 0000    49 44 30 31 39 3b                                   ID019;          
2020-07-09:06:11:26.070965: port_close called
2020-07-09:06:11:26.071021: ser_close called
Closed rig model 214, 'TS-2000 - no clients, will reopen for new clients'
2020-07-09:06:11:26.071185: client lock disengaged
2020-07-09:06:11:26.073151: Connection closed from localhost:58658

mdblack98, 2020-07-10:
What version of hamlib are you using?  The last sequence you have looks really funky.  It opens and closes the rig, then tries to send commands.

Mike W9MDB

rutja76, 2020-07-10:
Using gpredict (2.3-33-gca42d22-1) for Debian on Raspberry Zero. I think the disengage is triggered by FT-817 (second last series of commands) here:

2020-07-09:06:11:25.939921: rig_open called
2020-07-09:06:11:25.940047: client lock engaged
2020-07-09:06:11:25.940098: rig_close called
2020-07-09:06:11:25.940144: ft817: ft817_close called

After which TS-2000 follows and disengages.

mdblack98, 2020-07-10:
If your hamlib is very recent (more current than Jun 28) then please add this switch to the ft-817 rigctld
--set-conf=post_write_delay=50

See if that helps.

Mike W9MDB

rutja76, 2020-07-13:
The hamlib I'm using is older. But I'll update and see if it changes anything.

rutja76, 2020-08-19:
> If your hamlib is very recent (more current than Jun 28) then please add this switch to the ft-817 rigctld
> --set-conf=post_write_delay=50

I tested it with hamlib daily snapshot (20200819) and it seems to do the trick.