csete/gpredict #235: Gpredict (or hamlib) overrides manual VFO changes

rutja76, 2020-10-09:
Using Gpredict with the latest version of hamlib (4.0 rc2), when satellite tracking is engaged, a manual change of the frequency using the rig knob is overridden. E.g., if the downlink frequency is 145.740 before Doppler, this is not changed when turning the knob of the rig. Gpredict forces the frequency back to 145.740 plus/minus Doppler.

This happens in Windows and Linux, both with FT-817 and TH-D74 connected via Bluetooth. It works with TH-D74 if I change the cycle to 5000 ms or keeping 1000 ms and moving the knob very fast. With FT-817 it doesn’t work as the cycle doesn’t seem to influence the frequency update interval.

mdblack98, 2020-10-09:
What's the last version that worked?
Can you test the latest by cloning the master repository?
git clone https://github.com/Hamlib/Hamlib
Can you also please provide some debug?
rigctld -m 1020 -r /dev/ttyUSB0 -s 19200 -vvvvv -Z >&log.txt

And send me the log.txt file.

Mike W9MDB

rutja76, 2020-10-10:
I think it is related to the speed of the Doppler shift. If Doppler frequency changes slowly (e.g. satellite not visible) I can get the frequency to change manually moving the knob even with a cycle of 1000 ms. But if I chose 100 ms, no way, I can turn the knob as fast as I want, Gpredict overrides it with the original frequency plus/minus the Doppler shift. 

The log contains an example using TH-D74 with Hamlib 4.0~rc2 (Last commit was Mon Sep 07 17:34:04 2020 +0000 SHA=b7452d), windows built. Cycle is 100 ms and I turn the knob all the time. No changes in the frequency, it stays at 145.740.000 plus/minus the doppler.

Raffa OH8FKS
[TH-D74.txt](https://github.com/csete/gpredict/files/5359451/TH-D74.txt)

mdblack98, 2020-10-10:
Try the latest commit I just did.  Had to fix some VFO problems.
And try disabling caching -- that sounds like the cause for when you're doing 100ms polling as it defaults to 500ms.--set-conf=cache_timeout=0
Mike

rutja76, 2020-10-10:
How do I compile the latest commit? There is no configure file.

mdblack98, 2020-10-10:
First you need to do this:./bootstrap

rutja76, 2020-10-10:
Same behaviour also with the latest commit. Adding --set-conf=cache_timeout=0 helps a bit, but doesn’t solve the problem. With caching disabled it seems to work all the time doing 1000 ms polling. Going down to 100 ms, it works only if you spin the knob very fast. If you change like 20 kHz up or down, it overrides it and brings you back to the frequency you were before. Is anyway ok to disable caching?

mdblack98, 2020-10-10:
You can always run with caching turned off.  
I'll have to see if I can duplicate this problem.
Mike

mdblack98, 2020-10-10:
After you start rigctcld you can turn on "twiddling"
For example...to set twiddle to 1 second
rigctl -m  2 \set_twiddle 1
See if this solves the problem.  This was implemented quite a while ago but is not well known.
Mike