csete/gpredict #239: Gpredict received signal SIGFPE Arithmetic exception

sv1, 2020-11-19:
I am using gpredict 2.3.78-6add under Debian 10. From time to time gpredict is crashing with no apparent reason. Although this is not often it is a little annoying. 

Today I decided to compile it again and use gdb in case I was lucky:

```
[Thread 0x7ffff0a8b700 (LWP 21157) exited]
[Thread 0x7fffea60f700 (LWP 21156) exited]
[New Thread 0x7fffea60f700 (LWP 31909)]
[Thread 0x7fffea60f700 (LWP 31909) exited]

Thread 1 "gpredict" received signal SIGFPE, Arithmetic exception.
0x0000555555582852 in gtk_polar_view_create_track (pv=0x555555930670, obj=0x5555572748f0, sat=0x55555571f400)
    at gtk-polar-view.c:1561
1561	        if (!(i % tres))
(gdb)

```

But this is how far I can go!

I can provide current tle and ground station coordinates if this is necessary.

mdblack98, 2020-11-19:
And what are the values of i and tres at that point?  Might need a boundary check there.
Mike W9MDB

sv1, 2020-11-20:
`````
(gdb) p i
$1 = 1
(gdb) p tres
$2 = 0
(gdb) 
``````

mdblack98, 2020-11-20:
I'm guessing the solution is this...can't do modulus operation with zero.

if (tres != 0 && !(i % tres))

csete, 2020-11-21:
Thanks for catching this. I guess it is a bug somewhere else that tres is 0, but for now, I have added the check as suggested by @mdblack98.

closed 2020-11-21 by csete