csete/gpredict #242: Satellite footprint visualization

csete, 2020-12-19:
The branch [feature/coverage](/csete/gpredict/tree/feature/coverage) contains [pull request 223](#223) by @brookst, which makes the satellite footprints and names on the map optional. Instead of having these as a per-satellite on/off option, we could make these settings that apply to all satellites on the map and extend it to something like:

* Satellite names (enumeration):
  * Enabled (default): Satellite name is always shown
  * Disabled: Satellite name is never shown (exept in the tooltip)
  * Selected: Satellite name is only shown for the selected satellite
* Range circle (enumeration):
  * Enabled (default): Range circle is always shown
  * Disabled: Range circle is never shown
  * Selected: Range circle is only shown for the selcted satellite
* Footprint area, i.e. te area within the range circle (enumeration):
  * Disabled: No highlighting of the footprint area
  * Highlighted (default): Highlighted as today
  * Other options can be added later to visualize e.g. coverage of a constellation or how close a satellite is

GilbertGn, 2021-05-20:
Has this enhancement been released yet?

mykhal, 2023-05-30:
I agree that ground footprint toggle would be more useful if being a global (per module) option too, as one probably wants to hide footprints for large constellations.

(Btw, unfortunately, currently it's impossible to do that even in sub-optimal way by colors config, as Satellite color covers footprint border line color as well).

csete, 2025-12-26:
Both the satellite names, footprints, and markers can now be disabled on the map and polar views. On the map, the will still become visible if the satellite is selected, then hidden again if the satellite deselected.