csete/gpredict #246: support goocanvas 3

chenrui333, 2021-01-19:
👋 it looks like current gpredict 2.2.1 only supports `goocanvas 2`

> `configure: error: Gpredict requires libgoocanvas-2.0-dev` looks like we need to create `goocanvas@2`

relates to https://github.com/Homebrew/homebrew-core/pull/69285

csete, 2021-01-23:
As far as I can see, goocanvas3 has just been released a week ago, so yeah...

I will take a look and see whether it is just a version change or whether there is API/ABI breakage that requires changes in the code. If it is just a version change, we can probably easily support both version 2 and 3. However, in if breaks API or ABI, you should probably prepare to ship both version 2 and 3 and give applications time to migrate properly.

chenrui333, 2021-01-23:
Sounds good. Thanks @csete!

csete, 2021-02-20:
@chenrui333 As far as I could see from you patch (PR #251), Gpredict could build against GooCanvas 3 without changes to the code. So I have updated the configure.ac to check for GooCanvas 3 if GooCanvas 2 is not available. Could you please check the current HEAD and see if it works like this? Thanks.

LongnoseRob, 2021-03-21:
Just a short feedback,
I can confirm this fix on HEAD works, atleast on archlinumx-arm with goocanvas 3.0.0-1.
NO ssue with compiling with this version of goocanvas

csete, 2021-05-08:
Thanks, @LongnoseRob!

closed 2021-05-08 by csete