csete/gpredict #253: Rotator control with rotctld - Gpredict shows "ERROR" then segfaults

Aang23, 2021-02-07:
I have lately been using GPredict to control a rotator via rotctld.

Rotctld is started as a dummy rotator (rotctld -m1 -v), to which I connect GPredict with the default configuation. Then I ran other programs connecting on the same rotctld instance and at first, everything worked at expected and I was able to track a few satellites.

Once in while though, GPredict fails to control the rotator and shows the following, in order : 
[image]
[image]

From that point on, any attempt to connect back onto Rotctld fails, until I close the antenna control window, reopen, and reconnect, but most of the time doing so results in a segmentation fault.

GPredict was built from the latest Github sources, in fact, a few minutes before posting this issue to make sure before reporting.

Any idea what the problem could be? The issue is also observed with gpredict as the only client on Rotctld.

Thanks!

Aang23, 2021-05-22:
> Hi @Aang23 and sorry for the late reply.
> 
> The rotator interface should work also when other clients are connecting to rotctld. I suspect the problem might be somewhere else, for example, in the tracking logic because ELECTRO-L2 is a geostationary satellite, and predict still tries to track it using the SGP4 algorithm. Could you perhaps try and see whether the same problem occurs with other satellites that are not in geostationary orbit?

Hi and thanks for you reply! - Didn't get the notification so I only just saw it...
I did try with LEO satellites as well back then. The exact same issue occurred.

zaynpatel, 2024-02-28:
@Aang23 - Did you ever get this working? My current setup is:

- [Yaesu rotor controller](https://www.yaesu.com/indexVS.cfm?cmd=DisplayProducts&ProdCatID=104&encProdID=79A89CEC477AA3B819EE02831F3FD5B8)
- Hamlib script that's defining /dev/tty ports which the rotor is connected to
- GPredict to track satellites from this setup

I am getting a very similar message to yours where the azimuth/elevation blocks have: `read:ERROR`. I'm curious if you got anything working and what your setup looked like.
[image]

sdhombres, 2025-05-22:
It is due to the line 323, in file gtk-rot-ctrl.c (https://github.com/csete/gpredict/commit/b74d9bdf9fcfc27d7ea606d779a8f6d6f33a6cb7):

    /* try to parse answer */
    if (retcode)
    {
        if (strncmp(buffback, "RPRT", 4) == 0)
        {
            g_strstrip(buffback);
            sat_log_log(SAT_LOG_LEVEL_ERROR,
                        _("%s:%d: rotctld returned error (%s)"),
                        __FILE__, __LINE__, buffback);
            // retcode = FALSE; // COMMENT THIS LINE
        }

When we commented the line 323 and we recompiled Gpredict, the "engage" button remains selected when using the rotor controller GS-232 and we don't see "ERROR" in Az/El labels.

sdhombres, 2025-06-10:
We also noticed this bug with the latest version.
It seems to be due to the line 323, in file gtk-rot-ctrl.c (https://github.com/csete/gpredict/commit/b56d77d988d0fdb5fcedded11bc4e2721b3f2ecb):

    /* try to parse answer */
    if (retcode)
    {
        if (strncmp(buffback, "RPRT", 4) == 0)
        {
            g_strstrip(buffback);
            sat_log_log(SAT_LOG_LEVEL_ERROR,
                        _("%s:%d: rotctld returned error (%s)"),
                        __FILE__, __LINE__, buffback);
            // retcode = FALSE; // COMMENT THIS LINE
        }

When we commented the line 323 and we recompiled Gpredict, the "Engage" button remains selected when using the rotor controller GS-232 and the "Read:" are corrects.