csete/gpredict #254: Rotation question

vnnspace, 2021-02-18:
I am using rotator BIG-RAS/HR with controller MD-01.
I have a question maybe I am doing something wrong and you can help me.
In both conditions -180 -> 0 -> +180 or 0 -> 180 -> 360 happen same things. 
When the system is tracking satellite lets estimate like on picture 
[image]
As you can see on the initial stage of tracking Az was 1 degree after several seconds it becomes a 359 degree and system star to move full cycle to reach 359 degrees.... not continue moving by track ... make full cycle to the 3xx ... it is very weird ... I understand if system moving to the initial point like that it is ok but when you already tracking why it does like this ???
It can be several solutions as I think 
1) I am doing wrong tuning of GPREDICT.
2) I am doing wrong tuning of MD-01
3) I am done some wrong running of rotctld ( but I am not tuning it at all )

If possible please help me with it...
Also, it will really help me if somebody can provide a final manual for the Gpredict, because manual which I have ( download from here s from 2018 ) and some fields in setting are absent in the manual, and I am not sure what to write there?
Also, I can not find Parking position in Gpredict ( where I can change it  ? ) because 180:45 is not applicable to me.

Thank you very much. I am sorry for so many questions.

mdblack98, 2021-02-18:
From the documentation I have of the MD-01 it does not have a "direction".We simply tell it to move to an az/el and it does whatever it wants to get there.
Mike W9MDB

vnnspace, 2021-02-18:
I see.
Thank you it means I should tune MD-01 itself.

Ok.

closed 2021-02-18 by vnnspace