csete/gpredict #256: Building doesn't work when following README instructions

tomcass240, 2021-04-13:
When trying to build from source by following the instructions in the readme file, it says to extract the source and run ./configure. I can't find any file called "configure" so typing ./configure doesn't do anything but there is a file called "configure.ac" in the folder. I'm pretty new to Linux and after doing some reading I found out I needed to type "autoconf" instead of ./configure. Unless I did something wrong I think the README file might need updating.

mdblack98, 2021-04-13:
Actually "autoreconf -i" is the correct thing to do.
Mike W9MDB

davidfdzp, 2021-04-13:
I had the same issue and I read somewhere that if you are building directly from the git repository, you have to run
./autogen.sh instead of configure. You can pass the same options to the autogen.sh script as you would to the configure script.
Certainly, updating the README would nice.

csete, 2021-04-14:
@davidfdzp is correct and the info comes from the README ;)
https://github.com/csete/gpredict/blob/master/README#L65

I think what is confusing is that GitHub automatically generates tar.gz and zip source packages called "Source code", which are a copy of the git repository, hence they do not have the configure script. The official release packages have a version number in the file name, i.e. gpredict-2.2.1.tar.bz2, and these packages come with the configure script.

I think the README is correct, however, if anybody can think of a better or clearer way to write it, please let me know.

closed 2021-04-14 by csete

mdblack98, 2021-04-14:
I've got the same problem with hamlib....I wonder why we don't just include the configure file in the github project to avoid this problem.That would prevent people from needing all the additional tools to compile things.

Mike W9MDB

 
 

https://github.com/csete/gpredict/blob/master/README#L65

csete, 2021-04-17:
> 've got the same problem with hamlib....I wonder why we don't just include the configure file in the github project to avoid this problem.That would prevent people from needing all the additional tools to compile things.

I think the configure file would change quite often when different developers build on different machines and it would be confusing when to check it and when not to check it in.

mdblack98, 2021-04-17:
Yeah...I've decided against that one...it needs numerous other files to go with it.