csete/gpredict #257: Change button colour scheme

whattwood, 2021-04-17:
Changing the colour scheme a bit would be helpful to see at a glance if a toggle button is enabled. Right now it can be difficult to tell if a button (I mostly use them in the radio control window) has been clicked because it simply goes from a gradient grey to a solid grey.

Just my humble opinion - feel free to close the issue if this wouldn't be easily accomplished!

[image]

csete, 2021-04-18:
Hi @whattwood 
I fully agree with you, however, this is not an issue in gpredict but a problem with the theme/style you are using on your system. Gpredict uses a graphical toolkit that is integrated into the desktop environment and the look is controlled via the system settings. This is how it looks on my computer running XFCE desktop:

[image]

You can clearly see the difference between the T and L buttons.

whattwood, 2021-04-20:
Thanks,

 

I’ve tried changing a number of settings in the Windows 10 color scheme including some regedits, but can’t seem to find it. Will keep looking!

 

All the best,

Bill

VE6WK

 

 <https://user-images.githubusercontent.com/306178/115139857-a8764980-a034-11eb-89c3-f633402cbf20.png> ksnip_20210417-173404

csete, 2021-04-25:
Ah, okay, I thought it was on Linux. On Windows we might be able to improve it since the widget toolkit is included in the binary package. I will take a look for the next release.