csete/gpredict #259: Display all ground stations on the map

nabelekt, 2021-05-07:
Is there any way to display all of the ground stations on the map? If not, it would be great if that could be implemented.

csete, 2021-05-08:
Currently, a module can have only one ground station associated with it but this will change sometime in the future.

nabelekt, 2021-06-18:
Thanks. It would be helpful to be able to be able to have multiple ground stations for each module where one is selected as "active", and that one is used for the polar plot and info display, and the other "inactive" ones simply display on the map.

troygrosfield, 2022-11-29:
Hey @csete- Great product!  We're also very interested in the feature above.  Is this still on the future roadmap?  If so, any ideas on when it might be implemented as this will impact our decision to this software.