csete/gpredict #262: Solar Terminator Other Features Not Working

GilbertGn, 2021-05-20:
In regards to gpredict 2.2.2 or 2.3.33

1 - How do you get the Solar Terminator and Global shadow to display on the map view?
When I go to Preferences --> Map View
There is a list of items where the color can be changed. Included in the list is Solar Terminator and Global shadow. When I set either one to something like red and the opacity set to the maximum they don’t appear on the map view.

2 - The opacity/transparency slider appears not to work.

LongnoseRob, 2021-05-21:
The soalr terminator works, as can be seen, please note that this is gpredict built from the latest master of this repo

[image]
[image]

GilbertGn, 2021-05-22:
Now that is interesting. 
I have two different versions running in two different environments.
- On Fedora 33 I have Gpredict version 2.2.2 
- RPi 3B running the latest release of Buster the Gpredict version is 2.3-33

But when I look at the Map View settings the check box for the Solar Terminator is missing.

The question why is it missing on both of the instances of Gpredict I have?

What OS is yours running on?

[image]
[image]
[image]

LongnoseRob, 2021-05-22:
The screenshots were made on my PC with ArchLinux, gpredict compiled Freon the latest git sources.
Also have it running on ArchLinuxARM also bukt from the latest git sources.

GilbertGn, 2021-05-22:
Could point me to the source you compiled from?
I got the gpredict-2.2.1.tar.bz2 from Github and that Solar Terminator check box is missing in the source. So I'm not sure what source you used.

LongnoseRob, 2021-05-22:
there are two ways to get the lastest master of grpedict and compile it:
**1st way**
1. open a console and go to a directory where you would like to have the sources e.g. ~/builds
2. enter `git clone `https://github.com/csete/gpredict.git`
3. cd gpredict
4. enter `./autogen.sh`
5. enter `make` or 'make -j 4` (if you have a rather modern machine with more then 2GB of RAM)
6. enter `sudo make install`

**2nd way**

go to https://github.com/csete/gpredict
and cllick on the green code button and select zip-download
[image]
after downloaing the zip-file, unpack it to a folder of your choice and follow steps 4-6 from above.

PS I do not know which distribution you are using, and I did not include any steps to check if all requirements for compiling are installed.
If autogen.sh fails this is very probably due to missing packages required to compile or run gpredict/
Please also check the readme and the documentaton folder.

GilbertGn, 2021-05-22:
Well I had some success. I was able to get to build on the F33 system and the "Solar Terminator" option is there and works.

But on the Raspberry Pi with OS release buster   **./autogen.sh** gives this error.....  **./autogen.sh: intltoolize: not found**

I can't find the package it belongs to. 
Do you have any idea what the package might be?

LongnoseRob, 2021-05-23:
@GilbertGn great, glad that you got it compiled.

closed 2021-05-23 by GilbertGn