csete/gpredict #263: Satellite name in map view is rendered under the footprint

shuyuan-liu, 2021-05-20:
I'm on Windows 10 using Gpredict version 2.3.37 downloaded from Sourceforge. It looks like text labels in map view are rendered under (covered by) satellite footprints, making the text difficult to see if a footprint is set to a darker or more opaque colour. In the screenshot below the Area Coverage colour is #808080 with opacity 75%. Both satellite name labels and the groundstation label are obscured by the footprints.

[image]

Admittedly I wouldn't use such an opaque colour anyway because it doesn't look good, but still I think rendering text above footprints instead of below would improve readability.

shuyuan-liu, 2021-05-20:
Same on Linux, with Gpredict 2.3.103:
[image]

csete, 2021-11-13:
Thanks for the report. I agree we that should change this so that at least the satellite names are drawn above the footprint.