csete/gpredict #265: Add support for space-track.org in TLE Network Update

Marsfan, 2021-05-21:
[space-track.org](space-track.org) is a database maintained by the United States Space Force of all (non-classified) objects that the USSF is currently tracking. Access is allowed for the general public, but requires creating a login, therefore a simple URL cannot be used to allow network updating. However, there is a REST API to allow downloading data from the database. 

Adding support for using this API to access the space-track database will allow users to greatly expand the number of supported satellites in GPredict, without having to manually update the TLEs

csete, 2021-11-13:
It is on the TODO list for a future update

weather-tracker, 2022-01-10:
that will be nice

weather-tracker, 2022-01-11:
this looks helpful
https://www.space-track.org/documentation#howto-api_cplusplus
https://www.space-track.org/documentation#howto-api_curl