csete/gpredict #267: support for  800 x 480 resolution touchscreen

wuxx, 2021-06-15:
Hello, I'm using a raspberry pi with a 800 x 480 resoution 4.3 inch CSI touchscreen, it's tiny and I think good to be a outside station, we can use the touchscreen to open the gpredict , use the raspberry pi GPIO to control the radio and rotator, but unfortunately it seems the gpredict do not support the resolution 800 x 480 very well, for example I can not add a new ground station. which part of code can I try to fix it ? I would like to have a try
[image]

mdblack98, 2021-06-15:
Seems you can set up a virtual desktop and scroll...
https://raspberrypi.stackexchange.com/questions/95738/virtual-resolution-scrolling

Mike W9MDB

wuxx, 2021-06-16:
Thanks Mike! I update the framebuffer parameters in /boot/config.txt, now it works very well ! will close the issue
framebuffer_width=1600
framebuffer_height=960
[image]

closed 2021-06-16 by wuxx