csete/gpredict #269: Antenna Control Park

BogdanTheGeek, 2021-07-28:
I am working on a antenna rotator that acts as a `rotctld` server. I have gpredict controlling the antenna now and I was wondering if we could have a park button and maybe a parameter set field to specify the parking position?
`rotctl` has a command 'K' for park.
I could add it myself and do a PR if you don't have the time. The code base is very clean.

PS: thanks for the awesome tool.

BogdanTheGeek, 2021-08-06:
Also on the same line of thinking, could we get the peak amplitude from the fft sent over?

csete, 2021-11-13:
Sure we can add support for rotator parking. It would probably be useful in many setups. What do you think would be most useful:

1. A button as you describe?
2. A rotator setting to park the rotator whenever it is not tracking a satellite?
3. Both?

> Also on the same line of thinking, could we get the peak amplitude from the fft sent over?

Sorry, I don't understand this question.

F8DQL, 2023-03-28:
Hi, as it been implemented?
The option 3 would be great!

luzpaz, 2023-11-20:
@BogdanTheGeek any updates on this ?