csete/gpredict #270: Radio Control for a Module should save Devices and Cycle Times between its uses

tbondy760, 2021-08-08:
* Raspberry Pi, Buster
* 2.3-33-gca42d22-1 installed using build-a-pi (see https://github.com/km4ack/pi-build)

* I open Radio Control for a module
* It suggests FT817nd as the first device, I set it to FT891
* I set the second device to FT817nd
* I change the Cycle by a small number of tens of ms
* I select Engage, Track, and T on the dialog. Radios work as they should. After a time I close the dialog.
* I open the Radio Control for the same module again
* It has forgotten the devices and cycle time that I have set up, but has FT817nd as the first device and nothing as the second. Cycle is back to 1000 ms

Would love it if this also persisted between program executions.

tbondy760, 2021-08-08:
Looks like it uses `sudo apt install gpredict` according to https://github.com/km4ack/pi-build/blob/4475f256ea081069410f773e7ec18b00b72eb0eb/functions/additional.function#L577

csete, 2021-11-13:
The settings should be persistent. I don't remember though whether it is something that has been fixed since the version you mention.