csete/gpredict #280: James Webb Space Telescope tracking causes app to freeze

rt-bishop, 2022-01-03:
I've just found out that addition of JWST, Norad ID 50463, to the tracking list leads to a complete Gpredict freeze. I also checked my Look4Sat on Android and the n2yo.com website and both show crazy weird data. Do you possibly know the reason for this?

yashi, 2022-02-02:
Would you mind sharing how to reproduce?  I thought I successfully did the other day but when I tried this morning I just worked.  I could be with a certain TLE parameters?

rt-bishop, 2022-02-05:
@yashi, thanks for your response! You're right, I've just tried it and it doesn't crash, so I guess it was just some faulty TLE data.

closed 2022-02-05 by rt-bishop

yashi, 2022-02-07:
Glad it works for you.
I guess we have some issue in Gpredict. It's unfortunate that we can't reproduce anymore.

patyizsolt, 2022-06-26:
####