csete/gpredict #286: Error update transponder data

RF-TEC, 2022-04-11:
Gpredict v2.3-33-gca42d22-1 
Source : PPA
Dist : Gnu/Linux Mint 20

Hi

Please, how to resolve this error : 

Log :
```
2022/04/11 07:55:54|1|modes_update_from_network: Erreur lors de la récupération de https://db.satnogs.org/api/modes/?format=json (SSL peer certificate or SSH remote key was not OK)
2022/04/11 07:55:54|1|modes_update_from_network: Could not fetch frequency files from network
2022/04/11 07:55:54|1|trsp_update_from_network: Erreur lors de la récupération de https://db.satnogs.org/api/transmitters/?format=json (SSL peer certificate or SSH remote key was not OK)
2022/04/11 07:55:54|1|trsp_update_from_network: Could not fetch frequency files from network
```

csete, 2022-07-16:
Could be a temporary issue with the certificate. Are you still having problems?

RF-TEC, 2022-08-17:
Hello

Everything is in order.

Thank for your responsiveness and technical support.

I follow closely the evolution of the project, with great pleasure !

Best wishes

RFTEC

closed 2022-08-17 by RF-TEC