csete/gpredict #290: IC-9700 & rigctld & grpedict fail

dl2sba, 2022-05-26:
Hi All

I have problems running my IC-9700 (FW 1.31) with gpredict (gpredict-win32-2.3.37 or gpredict-win32-2.2.1) and hamlib (hamlib-w64-4.5~git or 4.4).

IC-9700 & gpredict is set according to https://github.com/csete/gpredict/blob/master/doc/notes/ic-9700.txt 

This is the command line for rigctld:

`            rigctld -m 3081 -r COM7 -s 115200 -T 127.0.0.1 -t 9700 -vvvvv`

When I click the ENGAGE-button the frequencies on the IC9700 flicker a few times, then the ENGAGE-button becomes unselected again.

I get this error log: [rigctld.log](https://github.com/csete/gpredict/files/8781435/rigctld.log)

Any clue what I did wrong?

Thanks in advance 

//Dietmar

mpippins, 2022-06-17:
I have this same problem, and a ticket open with Hamlib.  If you go down to Hamlib 4.0 or 4.1, it should work.  I can't get anything over 4.1 to work when the 9700 is in SAT mode.

dl2sba, 2022-06-18:
@mpippins Tnx will try this today - at least a glimmer of hope ;-)

dl2sba, 2022-06-19:
@mpippins I've loaded V4.1 of Hamlib and it seems to work now - thanks for this great hint!

closed 2022-06-19 by dl2sba

mpippins, 2022-08-28:
FYI, The new version has this fixed.  4.5 is worth upgrading to.  It doesn't lock up when moving the RX knob :)

dl2sba, 2022-08-28:
Tnx for info - will upgrade soon!