csete/gpredict #291: (BUG?) New satellites not add when update from local TLE

jake9wi, 2022-06-13:
When I update the TLEs from a local file satellites that are in the file but not in the GPredict database are not added.

- Win 10 x64 (21H2, 19044.1706)
- GPredict 2.2.1

csete, 2022-07-16:
Do you have the "Add new satellites to local database" selected in the preferences?

[image]

f6bvp, 2023-02-16:
I have the same issue with 2.3.115-0f3be version compiled from sources on a Linux Ubuntu system.
For example new Galileo satellites are not added in local galileo.cat file and thus Gpredict cannot compute the satellite orbits.
I am sudo when performing TLE update from Internet and there must be no write issue that could explain an update failure.