csete/gpredict #294: FT-857D Swaps Frequencies on frequency update

joshuajayg, 2022-07-06:
I'm using a OSX 11.6, hamlib 4.4, and gpredict 2.2.1.  I installed gpredict via macports then again via homebrew.  I also tried hamlib 4.0.  All the varieties have the same problem.

When controlling my 857D via CAT with gpredict, VFOa will try and update frequency but will instead take VFOb's frequency and usually, VFOb will double in frequency.  The problem only happens when VFOa tries to update frequency but it happens every time it tries to update.  Below are some snippets from hamlib's (rigctld) verbose output.

Startup:
`2022-07-06:21:29:43.568495: rigctld.c(579) Startup: rigctld -m 1022 -r /dev/tty.usbserial-AB0MVY9D -s 9600 -vvvvv -Z
2022-07-06:21:29:43.568549: rigctld Hamlib 4.4 Thu Dec 02 23:46:51 2021 +0000 SHA=5f8c4c
2022-07-06:21:29:43.568556: Report bugs to <hamlib-developer@lists.sourceforge.net>

2022-07-06:21:29:43.568576: rig_check_rig_caps: p1=0x110933180, p2=0x110939700, rig_model=0x110933180, macro_name=0x110939700
2022-07-06:21:29:43.568585: rig.c(359):rig_check_rig_caps return(0)
2022-07-06:21:29:43.568948: yaesu: initrigs4_yaesu called
2022-07-06:21:29:43.568959: rig_register called
2022-07-06:21:29:43.568966: rig_register: rig_register (1021)
2022-07-06:21:29:43.568985: register.c(225):rig_register return(0)
2022-07-06:21:29:43.568992: rig_register called
2022-07-06:21:29:43.568998: rig_register: rig_register (1027)
2022-07-06:21:29:43.569005: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569012: rig_register called
2022-07-06:21:29:43.569019: rig_register: rig_register (1010)
2022-07-06:21:29:43.569025: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569048: rig_register called
2022-07-06:21:29:43.569054: rig_register: rig_register (1005)
2022-07-06:21:29:43.569060: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569067: rig_register called
2022-07-06:21:29:43.569072: rig_register: rig_register (1006)
2022-07-06:21:29:43.569078: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569084: rig_register called
2022-07-06:21:29:43.569090: rig_register: rig_register (1007)
2022-07-06:21:29:43.569095: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569103: rig_register called
2022-07-06:21:29:43.569109: rig_register: rig_register (1039)
2022-07-06:21:29:43.569115: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569120: rig_register called
2022-07-06:21:29:43.569126: rig_register: rig_register (1009)
2022-07-06:21:29:43.569132: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569137: rig_register called
2022-07-06:21:29:43.569143: rig_register: rig_register (1020)
2022-07-06:21:29:43.569149: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569155: rig_register called
2022-07-06:21:29:43.569161: rig_register: rig_register (1001)
2022-07-06:21:29:43.569167: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569173: rig_register called
2022-07-06:21:29:43.569178: rig_register: rig_register (1022)
2022-07-06:21:29:43.569185: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569190: rig_register called
2022-07-06:21:29:43.569196: rig_register: rig_register (1023)
2022-07-06:21:29:43.569202: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569208: rig_register called
2022-07-06:21:29:43.569213: rig_register: rig_register (1011)
2022-07-06:21:29:43.569219: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569225: rig_register called
2022-07-06:21:29:43.569231: rig_register: rig_register (1015)
2022-07-06:21:29:43.569237: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569242: rig_register called
2022-07-06:21:29:43.569248: rig_register: rig_register (1013)
2022-07-06:21:29:43.569254: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569260: rig_register called
2022-07-06:21:29:43.569266: rig_register: rig_register (1014)
2022-07-06:21:29:43.569271: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569277: rig_register called
2022-07-06:21:29:43.569283: rig_register: rig_register (1028)
2022-07-06:21:29:43.569289: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569295: rig_register called
2022-07-06:21:29:43.569301: rig_register: rig_register (1031)
2022-07-06:21:29:43.569307: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569312: rig_register called
2022-07-06:21:29:43.569318: rig_register: rig_register (1016)
2022-07-06:21:29:43.569324: register.c(225):rig_register return(0)
2022-07-06:21:29:43.569330: rig_register called
[... 543 lines cut ...]
2022-07-06:21:33:25.160523: ft857_get_freq: called 
2022-07-06:21:33:25.160531: ft857: using cache (15 ms)
2022-07-06:21:33:25.160540: from_bcd_be called
2022-07-06:21:33:25.160551: rig_set_vfo: retcode from rig_get_freq = Command completed successfully
ft857: using cache (15 ms)
from_bcd_be called
rig_set_vfo: retcode from rig_get_freq = Command completed successfully
ft857: using cache (15 ms)
from_bcd_be called
from_bcd_be called
2022-07-06:21:33:25.160564: rig.c(1590):set_cache_freq return(0)
2022-07-06:21:33:25.160573: rig_set_vfo: return 0, vfo=VFOB, curr_vfo=VFOB
2022-07-06:21:33:25.160585: rig_set_vfo: elapsed=15ms
2022-07-06:21:33:25.160594: rig.c(2878):rig_set_vfo return(0)
2022-07-06:21:33:25.160605: rig_set_freq: elapsed=110ms
2022-07-06:21:33:25.160614: rig.c(2072):rig_set_freq return(0)
2022-07-06:21:33:25.160624: rig_set_split_freq: elapsed=111ms
2022-07-06:21:33:25.160633: rig.c(4035):rig_set_split_freq return(0)
2022-07-06:21:33:25.160643: rigctl_parse.c(2757):rigctl_set_split_freq return(0)
2022-07-06:21:33:25.160652: rigctl_parse: return#2 RPRT 0
2022-07-06:21:33:25.160693: mutex_rigctld: client lock disengaged
2022-07-06:21:33:25.160702: rigctl_parse.c(1767):rigctl_parse return(0)
2022-07-06:21:33:25.160711: handle_socket: doing rigctl_parse vfo_mode=0
2022-07-06:21:33:25.160720: rigctl_parse.c(660):rigctl_parse entered
2022-07-06:21:33:25.160728: rigctl_parse: called, interactive=1`

mdblack98, 2022-07-06:
Hopefully you're able to compile Hamlib?
If so please use the latest and see if it still has a problem.
git clone https://github.com/Hamlib/Hamlib.git

Mike W9MDB

mdblack98, 2022-07-06:
Also...try and use the --vfo option with rigctld.
Mike W9MDB

joshuajayg, 2022-07-06:
I can compile hamlib from source; that's how I got 4.0.  I'll try with the latest version.  I tried --vfo and -o.  The radio does not respond and the output of rigctld stops after... 

```
2022-07-06:22:00:20.498801: rig_parse_vfo called
2022-07-06:22:00:20.498806: rig_parse_vfo: '1' not found so vfo='None'
```

mdblack98, 2022-07-06:
I need to see your command line.

joshuajayg, 2022-07-06:
`blah_blah@blahblah ~ % rigctld -m 1022 -r /dev/tty.usbserial-AB0MVY9D -s 9600 -vfo -vvvvv -Z `
and
`blah_blah@blahblah ~ % rigctld -m 1022 -r /dev/tty.usbserial-AB0MVY9D -s 9600 -o -vvvvv -Z `

joshuajayg, 2022-07-06:
Well, Mike, Hamlib 4.4 is the latest version which I have installed and it works fine with other programs but I'm unable to compile it myself.  It doesn't come with config  or Makefile.in and I'm having trouble with automake.  Sooo...  compliling myself isn't happening today.

mdblack98, 2023-03-13:
Try the tar file from here
https://n0nb.users.sourceforge.net/

joshuajayg, 2023-03-14:
Sorry, I don't have the radio anymore so I can't test this.