csete/gpredict #321: Gpredict 2.3-72-gc596101-3 does not update galileo.cat

f6bvp, 2023-02-13:
Installed Linux Ubuntu gpredict 2.3.72 for Galileo satellite tracking with F4KLO radiotelescope antenna. 
Updating satellite TLE files from Internet does not update galileo satellites list galileo.cat and consequently new TLE galileo satellites are not included during update.
However TLE update is ok when performed from local file after downloading http://www.celestrak.com/NORAD/elements/galileo.txt

73 de Bernard, f6bvp / ai7bg
http://radiotelescope-lavillette.fr
(Site translation with Google translate on site)

f6bvp, 2023-02-13:
Windows 2.3.37 Gpredict version is performing Galileo TLE updates correctly from Internet.

mdblack98, 2023-02-13:
Looks like this was fixed in July 2022
commit 2103b2c4a6aadbbdad326f2a9d1de49e6032f094
Author: Sean Svihla <Sean.Svihla@lasp.colorado.edu>
Date:   Tue Jul 26 14:27:16 2022 -0600

    Changed top-level domains of CelesTrak TLE URLs
    
    Change CelesTrak URL extensions from .com to .org.
    Change CelesTrack and AMSAT protocol from http to https.

closed 2023-02-13 by f6bvp