csete/gpredict #328: creating new modules fail

lid-k, 2023-03-23:
Hello, I’m setting up Gpredict to run with SDSharp on Windows 10 (using these instructions [Weather Satellite Tracking Software Setup - YouTube 1](https://www.youtube.com/watch?v=bYpTsersKX8&ab_channel=TechMinds))
When I open the Gpridict software evrything seems ok until I press Create New Module or Configure, I get this error:

Assertion failed!
Program:
C:\SDR\gpredict-win32-2.2.1\gpredict-win32-2.2.1\gpredict.exe
File: …/src/strnatcmp.c, Line 123

Expression: a && b

Does anyone have any idea what this may refer to?

Thanks!
Lid

mdblack98, 2023-03-23:
That means a NULL value got passed into that function.

Try this version
https://github.com/Jansemar/Gpredict_win32

Mike W9MDB

lid-k, 2023-03-23:
Hi Mike, thank you for your quick advice! This other version hasn't worked either sadly

lid-k, 2023-03-23:
Fixed! When I reinstalled the second time the software was still connected to an older folder in a User directory so I've deleted that one and started from scratch in Local Disk (not user folder), and it now works.
Thank you for your help :)

closed 2025-12-27 by csete