csete/gpredict #329: Start Gpredict fullscreen

StellarStorm, 2023-03-26:
Hi @csete 

First - thanks for this great program! It's been one of the first things I've installed on any new computer since at least 2016! 😄 

Is it possible to start Gpredict in full screen mode automatically? I've repurposed an old Surface Pro without a keyboard, and I normally connect via ssh and start Gpredict from the terminal. I'm able to switch the display to fullscreen with a mouse but it would be nice to start fullscreen and not need any further interaction.

I found this [old discussion](https://sourceforge.net/p/gpredict/mailman/message/34386965/) but it doesn't look to have been updated. I also took a wild guess and tried `GTK_SAT_MOD_FULLSCREEN=1 gpredict` but that had no visible effect.

Cheers!

luzpaz, 2023-11-20:
Is there a solution for this ?

csete, 2023-12-24:
I have added a command line option --fullscreen to start gpredict in fullscreen mode. This is just a quick fix that will put the entire application into fullscreen mode and not just the active module.

Is this sufficient?