csete/gpredict #332: Crash when satellite EL=-0.00 at AOS in 00:00

emmanuelCNRS, 2023-05-22:
Dear Gpredict Team,

I would like to submit several crash of Gpredict when AOS is 00:00 and antenna EL is -0,00°.
Gpredict run with rotctld.
Gpredict version is win32-2.3.37 (on Windows7).

In attached pictures, two crash examples when AOS on UVSQ-Sat and INSPIRE-Sat7.
Could it be fixed with parameters adjustment ?

Best regards.
Emmanuel - F4IXT
From UVSQ-Sat & INSPIRE-Sat7 Ground Station ;-)

[image]
[image]

emmanuelCNRS, 2023-09-28:
Don't know if it could help, hereafter an exemple of the many crash reports saved by Windows:
[Report-wer.txt](https://github.com/csete/gpredict/files/12749824/Report-wer.txt)

Best regards.
Emmanuel

mdblack98, 2023-09-28:
#1 What do you have  for your pass conditions?
#2 Please run rotctld with this added:  -vvvvv >log.txt 2>&1
Then we can make sure it's no a  rotctld problem.

emmanuelCNRS, 2023-09-29:
Hi Michael,
Thanks for your comment, hereafter the pass conditions:
[image]

And hereafter the rotator configuration:
[image]

Rotctl is usualy used with:
_"C:\Program Files (x86)\hamlib-w64-4.4\bin\rotctld" -vvvv -m 603 -r COM6 -s 4800 -t 4533 -C timeout=2000_

Call to Rotctl have been modified has suggested (_-vvvvv >log.txt 2>&1_), hereafter the beginin of the log:
[log_20230929.txt](https://github.com/csete/gpredict/files/12763282/log_20230929.txt)

I will post the updated log as soon as next crash.

Thanks for your help :-)
Best regards,
Emmanuel

emmanuelCNRS, 2023-10-02:
Hi Michael,
I have captured a crash in the log (if needed I can share the entire file on an ftp):
[log_20231002_Extract.txt](https://github.com/csete/gpredict/files/12781815/log_20231002_Extract.txt)

At line 398122 you can find this: 
```
fscanf: Invalid argument
fscanf: parsing '
' with '%c'
Connection closed from ::ffff:127.0.0.1:61324
Connection opened from ::ffff:127.0.0.1:51123
```

Best regards,
Emmanuel

mdblack98, 2023-10-02:
Try the latest Hamlib -- is has better debug info and hopefully fixes this problem too.

gpredict is sending bad  data for some reason.

Tracking the issue here in Hamlib
https://github.com/Hamlib/Hamlib/issues/1351

Mike W9MDB

emmanuelCNRS, 2023-10-03:
Hello Mike,
Thanks, I have updated hamlib to "w64-4.5.5" version.
Waiting now for next satellites pass.

Best regards,
Emmanuel

mdblack98, 2023-10-03:
Latest Hamlib is here...

https://n0nb.users.sourceforge.net/

Mike W9MDB

emmanuelCNRS, 2023-10-06:
Hello Mike,
Thanks, I have installed "**hamlib-w64-4.6~git-20231003-c9344070**".

Gpredict has also crashed with "hamlib-w64-4.5.5", hereafter the rotctld log:
[log_20231006a - Beginning of file.txt](https://github.com/csete/gpredict/files/12832020/log_20231006a.-.Beginning.of.file.txt)
[log_20231006b - End of file.txt](https://github.com/csete/gpredict/files/12832023/log_20231006b.-.End.of.file.txt)
With a end like this:
```
write_block(): TX 3 bytes, method=2
0000    43 32 0d                                            C2.             
read_string_generic called, rxmax=32 direct=1, expected_len=1
read_string_generic(): RX 15 characters, direct=1
0000    41 5a 3d 30 39 32 20 20 45 4c 3d 30 30 30 0d        AZ=092  EL=000. 
gs232b_rot_get_position: (az, el) = (92, 0)
rot_get_position: got az=92.00, el=0.00
fscanf: Invalid argument
fscanf: parsing '
' with '%c'
Connection closed from ::ffff:127.0.0.1:54638
```
Waiting for next satellites pass,
Best regards.
Emmanuel

mdblack98, 2023-10-06:
Try tonight's build -- will be dated 20231007
It has more debug info so we can see what gpredict is sending.
Mike W9MDB

emmanuelCNRS, 2023-10-09:
Hi Mike,
Thanks, nightly build downloaded today and just installed. 
Begining of log is:
`rotctld, Hamlib 4.6~git 2023-10-08T22:18:16Z SHA=e68b6f 64-bit`

Emmanuel

emmanuelCNRS, 2023-10-12:
Hello Mike, 
Gpredict has crashed yesterday with the nightly built "Hamlib 4.6~git 2023-10-08T22:18:16Z".

The beginning of rotctld log is:
[log_20231011b - Beginning of file.txt](https://github.com/csete/gpredict/files/12883604/log_20231011b.-.Beginning.of.file.txt)

And the end with the unexpected gpredict crash is:
[log_20231011c - End of file.txt](https://github.com/csete/gpredict/files/12883616/log_20231011c.-.End.of.file.txt)

Best regards,
Emmanuel

mdblack98, 2023-10-12:
Are you compiling gpredict yourself?
I'd like to provide a patch for gpredict to figure out why it is sending a NULL byte.

emmanuelCNRS, 2023-10-13:
Hello Mike, 
No, I'm using the already compiled EXE available on https://github.com/csete/gpredict/releases
Version is "gpredict-win32-2.3.37".
If you send me a new EXE I can launch it in our ground station.

Best regards,
Emmanuel

emmanuelCNRS, 2023-10-18:
Hi Mike,
Please find hereafter some other log of rotctld (Hamlib 4.6~git 2023-10-08T22:18:16Z) with the unexpected closure of Gpredict.
[log_20231018_Extract.txt](https://github.com/csete/gpredict/files/12995314/log_20231018_Extract.txt)

The last one, at the end of the file, is not EL=000. But probably same problem since EL=179 is the opposite position (gpredict EL configured with EL in [0° ; 180°] range).

I have not recorded gpredict internal log, will be done next time, but should be something like this previous one:
```
2023/05/22 07:36:05|1|get_current_pass: Returning a pass for INSPIRE-SAT 7 that ends before the seeded time.
2023/05/22 18:21:58|1|get_current_pass: Returning a pass for UVSQSAT that ends before the seeded time.
```

Do you know where does come from "**that ends before the seeded time**" trace in gpredict?

Best regards,
Emmanuel

mdblack98, 2023-12-19:
Please try this package
https://www.dropbox.com/scl/fi/19300vq5wdgwg7vrywyrp/bin.zip?rlkey=j8tsgk0djg342y0yi9ss2j4e2&dl=0