csete/gpredict #334: Zenith Satellite Rotator Tracking

SMProgrammer, 2023-08-01:
I am using GPredict 2.3.37, hamlib 4.5.5, SPID MD01, rotctld, and a Big-Ras rotator, on a Windows 10 64 bit computer.

The commands sent to the rotator for near satellite passes near the zenith do not seem to be working properly. When a satellite passes near the zenith (an elevation angle of 90 degrees) with satellite tracking on and the rotator engaged, the rotator does a full 180 degrees azimuth turn instead of making the elevation angle go past 90 degrees. If the elevation angle was in the range of 90-180 degrees before the satellite, going through the zenith, the elevation angle sent to the rotator will stay above 90 degrees, and if the elevation angle was in the range of 0-90 degrees, the elevation angle will stay lower than 90 degrees. This full 180 degree azimuth turn takes time for the rotator to perform so the rotator does not track the satellite for a while. These high elevation angle passes are critical for communicating with a Cubesat so it is important that it is tracked consistently. I had the settings for the rotator having a maximum elevation angle of 180 degrees as shown in the image, but that did not make a difference. I can use GPredict to manually control the rotator and set it past 90 degrees in elevation, but GPredict does not use the full 180 degree elevation angle capability when tracking the satellite automatically.
[image]

https://github.com/csete/gpredict/assets/90346350/b66fc0c3-8218-46ee-ac08-1db61ec1d487

mdblack98, 2023-08-01:
Can you please test the azimuth zenith using rotctl instead of rotctld/gpredict?

Try 10 degree increments and see how it behaves.

I show max azimuth on the SPID  controllers is 540 degrees.

Mike W9MDB

https://github.com/csete/gpredict/assets/90346350/b66fc0c3-8218-46ee-ac08-1db61ec1d487

SMProgrammer, 2023-08-02:
Hi,
I tried ten-degree increments, and the same problem occurs. On rotctl I can move the elevation angle the full 180 degrees and the azimuth the full 360 degrees.