csete/gpredict #335: How to turn off satellite ground visibility circles and satellite names

olav6, 2023-08-11:
Hello! I've made a new github account just to ask this question: 

I use Gpredict to see all of Starlink satellites in GPredict but when I do that, I get the result as shown in the attached image. It's a bit confusing. How do I turn off the satellite ground visibility circles and satellite names off? I'd like to see the satellites as dots instead. 

[image]

Thanks!

olav6, 2023-08-11:
I see that this has been asked before and that it's called ground footprint, not "satellite ground visibility circles". 

https://github.com/csete/gpredict/issues/242#issue-771374438

olav6, 2023-08-11:
I thought you guys might get a kick out of this. 

After my last post, I realized I didn't have all of the Starlink satellites loaded up so I added all of them. One at a time. That took me a while. Being able to multi-select the satellites would have been greatly appreciated here. It also took gpredict a while to chew on that many satellites. About 15 minutes or so.  Here is what the result looks like. 

The Map View. 
[image]

The Sky View.
[image]

I do not know how to get the total number of satellites loaded in gpredict. According to this webpage, https://celestrak.org/NORAD/elements/table.php?GROUP=starlink&FORMAT=tle , there are 4,550 Starlink entries so I am assuming that is how many I have loaded up in gpredict. Yes, gpredit is a little laggy. It's not too bad with the refresh rate set to 60 seconds. I thought you guys might like to know that gpredict can handle this many satellites. 

So... can we have the option to have the satellites be represented as a dot in the Map View? :-D

csete, 2023-12-24:
Thanks for the info @olav6. Tracking so many satellites at the same time wasn't an issue when I initially wrote gpredict 20 years ago. I will see what I can do to improve, both the tracking and the visuals :-)