csete/gpredict #336: "METEOR M2-3" data is missing

gvimlag, 2023-08-28:
After "Update TLE data from network", the "METEOR M2-3" data is missing.

From: www.rtl-sdr.com/meteor-m2-3-now-in-orbit-and-transmitting-weather-images/
       "June 27, 2023 the latest Meteor M2-3 satellite was successfully
        launched on a Soyuz-2 and has been reported to be already transmitting
        LRPT images of the earth."

Please add this new satellite to your TLE database.

Thank you.

KaComet, 2023-08-29:
I can see METEOR M2-3 on my Gpredict. It is shown as METEOR-M2 3. Please ensure that you have updated your TLEs by going to Edit->Update TLE data from network. 

Sometimes satellite names will be a bit different across publications and LTEs. In some cases, I've seen several satellite use more than one name. To be safe, search by the satellite's NORAD ID. In the case of METEOR M2-4, the NORAD ID is 57166.

gvimlag, 2023-08-30:
OS: Debian 11, KDE
Gpredict: 2.3-72-gc596101-3

Hello KaComet,
Thank you for the response.

I did a fresh install of Gpredict and updated its TLE database.
```
<new install of Gpredict>
rm -rf ~/.config/Gpredict    #start completely clean (remove all old data)
Gpredict
Edit -> Update TLE data from network
Module menu -> Configure
```

I searched for "57166" and "METEOR"; but the "METEOR-M2 3" satellite (or similar) is not present.

Below is a screen shot of my TLE sources:
[image]

Do I need to add/update one of these sources?

Thanks for the help.

ploeffler, 2023-10-29:
confirm that
pulled current version form github and compiled on Ubuntu, but same problem

sv1, 2023-11-24:
Hello @gvimlag

you are using a pretty old version of Gpredict. Meanwhile Celestrak has changed the domain from .COM to .ORG.

If you can not upgrade to a newer version of Gpredict (Debian has 2.3-115 in Bookworm) you can manually change the TLE sources from celestrak.com to celestrak.org and perfom again an "Update from network"

Happy tracking!

sv1, 2023-11-24:
@ploeffler check if you have the same problem as @gvimlag otherwise probably network issue??? Give some more info.

ploeffler, 2023-11-24:
I guess there was some data corruption, with the update TLE I installed right now it works. 

thanks for investigating

luzpaz, 2023-12-25:
Is this solved ?

ploeffler, 2023-12-25:
as for me: yes

gvimlag, 2023-12-26:
Yes, the problem is solved.

For future readers, to fix this problem:

1. Edit: Edit->Preferences->General->TLE Update->TLE sources:
2. As sv1 instructed: change all ".com" to ".org".
3. Do *NOT* use "https".  Only use "http".  Otherwise, you will get SSL certificate errors which will prevent satellite TLE data from downloading.

luzpaz, 2023-12-26:
> Do NOT use "https". Only use "http". Otherwise, you will get SSL certificate errors which will prevent satellite TLE data from downloading.

Should a new ticket be opened for this issue ?

gvimlag, 2023-12-26:
> Should a new ticket be opened for this issue ?

Done.  See: https://github.com/csete/gpredict/issues/344

closed 2026-03-22 by csete