csete/gpredict #338: TEVEL-6 sat missing

shepherd007, 2023-11-27:
**Gpredict version:** 2.3.37
**OS:** Windows 10

I cannot select TEVEL-6 as a satellite to track. There are TEVEL-1 to TEVEL-5 and TEVEL-7 to TEVEL-8 available.
I already tried updating TLE file from network - and uploading locally TLE file downloaded from:
http://celestrak.org/NORAD/elements/gp.php?GROUP=amateur&FORMAT=tle

Anyone has seen this issue?

mdblack98, 2023-11-27:
You don't show this list in the Configure page?

Mike W9MDB

 
 

http://celestrak.org/NORAD/elements/gp.php?GROUP=amateur&FORMAT=tle

shepherd007, 2023-11-27:
When I enter "Configure" there is no TEVEL-6 satellite available (see attached screenshot):

[image]

mdblack98, 2023-11-27:
See if you have this file in your GPredict data store

C:\Users\mdbla\Gpredict\satdata> more .\50999.sat
[Satellite]
VERSION=1.1
NAME=Tevel-6
NICKNAME=Tevel-6
TLE1=1 50999U 22002R   23331.15431418  .00038335  00000+0  10890-2 0  9998
TLE2=2 50999  97.4133  38.4015 0006963  16.9632 343.1843 15.36058328103712
STATUS=0

When I enter "Configure" there is no TEVEL-6 satellite available (see attached screenshot):

configure.JPG (view on web)

closed 2023-11-27 by shepherd007