csete/gpredict #343: Confusing release versions across the downstream package ecosystem

luzpaz, 2023-12-25:
Here's a snapshot of the current downstream distribution of gpredict in the package ecosystem:

[image]

Ignoring all obsolete and unique entries, notice that there is a mixture of 2.2.1, 2.3, 2.3.x 

But we don't see any mention of 2.3 in Github, the release version shows v2.2.1. The [NEWS](https://github.com/csete/gpredict/blob/master/NEWS) file shows 2.2 (no mention of the minor version 2.2.1 or 2.3).

Would it be possible to bring more uniformity to this situation ? TIA!

csete, 2023-12-26:
2.2.1 is the latest stable release.
2.3-x are development snapshots from git, and because 2.2.1 is very old now, some distributions have started distributing these snapshots.

luzpaz, 2023-12-27:
Thanks for the reply. Would it be possible then to keep the NEWS file up to date? (Adding 2.2.1)

csete, 2025-12-27:
There is now a new version 2.4
I hope we can avoid such confusion moving forward :)

closed 2025-12-27 by csete