csete/gpredict #344: "https" prevents TLE data from downloading

gvimlag, 2023-12-26:
OS: Debian 11, KDE
Gpredict: 2.3-72-gc596101-3

Edit->Preferences->General->TLE Update->TLE sources:
If you change "http" to "https", you get SSL certificate errors which prevents satellite TLE data from downloading.

csete, 2023-12-26:
I guess the certificates are missing... Is the package ca-certificates installed?

gvimlag, 2023-12-26:
> Is the package ca-certificates installed?

Yes.  ca-certificates version 20210119 from the Debian 11.8 repository.

csete, 2025-12-27:
I hope this issue has been resolved in the meanwhile.
I just tried on a newly installed Debian 13.2 computer with build dependencies installed exactly as in the readme, and both TLE and transponder data download works normally.