csete/gpredict #347: Doppler control bug w/ IC-910

sudobob, 2024-01-28:
This issue is closely related to the issue with the 9700  - https://github.com/csete/gpredict/issues/181

I'm using 

Gpredict 2.2.2
rigctl Hamlib 4.5.5 Apr 05 11:43:08Z 2023 SHA=6eecd3

Gpredict radio configuration:
[image]

Movie of display behavior
https://i.imgur.com/aISSRul.mp4

I read that this was apparently fixed for the IC-9700 but apparently it hasn't for the IC-910? 

Here is a rigctld -vvvv -Z output:

[ic910-doppler-issue-trace.txt](https://github.com/csete/gpredict/files/14076998/ic910-doppler-issue-trace.txt)

..Or should this issue be taken up with hamlib?

Thanks for all the great software!

mdblack98, 2024-01-28:
Could you please add one more v to the debug output?

This is a hamlib problem.

Mike W9MDB

image.png (view on web)

https://i.imgur.com/aISSRul.mp4

sudobob, 2024-01-28:
I ran the -vvvvv.. I was getting different and inconsistent behavior and was noticing alot 'rig rejected commands'. My IC-910 baud rate was set to Auto. I set it to 19200 and also explicitly started rigctl with --serial-speed=19200 and now things seem more consistent. 

You can close this issue. Thanks.

mdblack98, 2024-01-28:
You mean it's working now?

wkitty42, 2024-01-29:
On 1/29/24 8:07 AM, Bob Coggeshall wrote:
> Yes. So far as I have tested It looks like the erratic behavior was all due to 
> serial line errors. If I do leave it for a long time they crop up.

have you tried 9600? it is slower but generally also less susceptible to 
noise... i'm curious if they still crop up after a while, the same, longer, or 
shorter than what you are seeing at 19200... or maybe they are in time with 
other electrical events... i have other thoughts but they are more on the serial 
side than gpredict so i'll stop here ;)