csete/gpredict #351: windows build, changing frequency using the mouse scroll wheel

ando274, 2024-05-19:
have been using gpredict 2.3.37 on Widows 10 successfully with an IC-9700, It all works very well. The issue is the use of the mouse scroll wheel on the “rig control frequency widget” does not work, the tool tip mentions “Use mouse buttons or wheel to change”.

Investigating the issue, I could see the code was not responding to mouse wheel events, also all the code required to deal with the events was in the source code of gtk-freq-knob.c

Enabling the scroll wheel mask with the line gtk_widget_add_events(knob->[i], GDK_SCROLL_MASK), at line 361 fixed the issue.

I have found using the wheel instead of the buttons to spot my down link signal as well as tuning other stations that are off frequency, faster and more convenient than the buttons.

I am not sure if this effect is particular to windows builds, as I have not tested on a Linux machine.

I used msys64 on Windows 10 to build and tested 32 bit and 64 bit versions, I have not found any problems.

curtiswilliams, 2024-06-27:
Can you share that file I’ve tried for weeks to get Mingw64 to compile for windows 11 and I’m failing miserably.

ando274, 2024-06-28:
https://1drv.ms/u/s!AioHOutp0it5i9wT0JoUSlBttvStoA?e=cKHDD6

Hi Curtis, the link is to a 64-bit version which started from the latest source about a month ago. It seems to work well for me, 

To build I used MSYS2, and the package installer to load the mingw64 dependencies and used the gprdict/w32 folder make file. The make file needed to be modified as it failed at the windows resource file and the paths needed adjustment.

I have not tested the version on Windows 11

cheers

csete, 2024-09-17:
There is a patch for this in issue #362 which I think fixes this in general. To be honest, I don't remember whether this was never implemented or just stopped working as the GUI toolkit evolved over the years.