csete/gpredict #352: Rotor Control window displays with missing knob(s) and crashes when tracking.

rakar, 2024-06-04:
I am new to GPredict and wanted a new compile (post 3.2.37) in order to avoid wrap around tracking. I followed directions for installing MSYS2 and with only minor fussing was able to compile for windows. The program appears generally in good order (some config defaults seem off). The main issue is that when I open Rotor Control, the Az/El knobs don't show up (Az is always missing, El is usually missing). I've tracked this down to the length of the b[7] array definition, which I believe should be 8. Just to be extra safe I set it to b[9] (not familiar with g_ascii_formatd and termination etc.) and it appears to work as intended on initial opening and tracking updates. 

```
static void gtk_rot_knob_update(GtkRotKnob * knob)
{
    gchar           b[7];  //<-------- Here
    gchar          *buff;
    guint           i;

    g_ascii_formatd(b, 8, "%6.2f", fabs(knob->value));

    /* set label markups */
    for (i = 0; i < 6; i++)
    {
        buff = g_strdup_printf(FMTSTR, b[i]);
        gtk_label_set_markup(GTK_LABEL(knob->digits[i + 1]), buff);
        g_free(buff);
    }

    if (knob->value < 0)
        buff = g_strdup_printf(FMTSTR, '-');
    else
        buff = g_strdup_printf(FMTSTR, ' ');

    gtk_label_set_markup(GTK_LABEL(knob->digits[0]), buff);
    g_free(buff);
}
```

ando274, 2024-06-05:
Well spotted, I had also seen that issue, but only on windows 64-bit builds, 32-bit builds worked fine. I have built a 64-bit version including your suggested mods and it seems to work perfectly. I had spent a number of hours trying to solve that bug, so thanks for that.

rakar, 2024-06-05:
You're welcome. Happy to help. Unfortunately the current version still seems to take the long way around when tracking so I'm looking at that next.