csete/gpredict #354: ICOM 2730a > Hamlib -> Gpredict , Can't get it to change frequency based on doppler shifts. But I can manually change the Downlink frequency.

curtiswilliams, 2024-06-13:
I am running a .bat : rigctld -vvvvv -m 3072 -r COM1

It's not working with adjusting the frequency as GPredict makes doppler shift adjustments. If I change the radio control Downlink frequency it gets sent and changes radio, but if I choose to Track then the doppler frequency is not changing on radio. 

Can you identify any issue or suggest a better .bat command that may assist in getting the frequency to consistently change with what GPredict is changing the frequency to?

In GPRedict I have Host localhost, Port 4532, Radio type I've tried all choices, PTT Status I've tried all settings, also VFO Up/Down same thing. No matter what I do with settings I can't resolve the issue.

[image]

LOG SEGMENT

C:\Users\garmi\OneDrive\Desktop\AntRunner-master\software\hamlib-w64-4.5git\bin>rigctld -vvvvv -m 3072 -r COM1
rigctld.c(626) Startup: rigctld -vvvvv -m 3072 -r COM1
rigctld Hamlib 4.5git Tue May 17 22:03:48 2022 +0000 SHA=468dd1
Report bugs to [hamlib-developer@lists.sourceforge.net](mailto:hamlib-developer@lists.sourceforge.net)

Max# of rigctld client services=32
rig_check_rig_caps: p1=000000005C104E60, p2=000000005C10B3D8, rig_model=000000005C104E60, macro_name=000000005C10B3D8
initrigs4_icom: _init called
rig_init(496): 0000000000F2AFFC rs->comm_state==0?=0
rig_init: rig has VFO_SUB
rig_init: rig has VFO_MAIN
1:icom.c(613):icom_init entered
icom_init: done
1:icom.c(695):icom_init returning(0)
main: twiddle=0, uplink=0, twiddle_rit=0
1:rig.c(743):rig_open entered
rig_open: cwd=C:\Users\garmi\OneDrive\Desktop\AntRunner-master\software\hamlib-w64-4.5git\bin
rig_open: C:\Users\garmi\OneDrive\Desktop\AntRunner-master\software\hamlib-w64-4.5git\bin/.hamlib_settings does not exist
serial_open: COM1
serial_setup: tcgetattr
serial_setup: cfsetispeed=19200,0x000e
serial_setup: cfsetospeed=19200,0x000e
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: Handshake=None
serial_setup: tcsetattr TCSANOW
read_string_generic called, rxmax=4095 direct=1
2:rig.c(7055):async_data_handler_start entered
async_data_handler_start: async data support disabled
2:rig.c(7060):async_data_handler_start returning(0)
rig.c(241):add_opened_rig returning2(0)
rig_open: 0000000000F2AFFC rs->comm_state==1?=1
2:icom.c(944):icom_rig_open entered
icom_rig_open: IC-2730 v20220326.0
3:icom.c(748):icom_get_usb_echo_off entered
4:frame.c(425):icom_transaction entered
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
5:frame.c(138):icom_one_transaction entered
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1
write_block(): TX 6 bytes, method=2
0000 fe fe 90 e0 03 fd ......
read_string_generic called, rxmax=200 direct=1
read_string_generic: minlen now 6
read_string_generic(): RX 6 characters, direct=1
0000 fe fe 90 e0 03 fd ......
icom_one_transaction: frm_len=6, frm_len-1=fd, frm_len-2=03
5:frame.c(403):icom_one_transaction returning(0)
4:frame.c(455):icom_transaction returning(0)
icom_get_usb_echo_off: ack_len=1
read_string_generic called, rxmax=200 direct=1
read_string_generic: minlen now 11
read_string_generic(): RX 11 characters, direct=1
0000 fe fe e0 90 03 00 00 62 45 01 fd .......bE..
icom_get_usb_echo_off: USB echo on detected, get freq retval=11
3:icom.c(777):icom_get_usb_echo_off returning(0)
icom_rig_open: echo status result=0
icom_rig_open: echo status known, getting frequency
rig_get_freq(1837) called vfo=currVFO
vfo_fixup:(from rig_get_freq:1843) vfo=currVFO, vfo_curr=None, split=0
vfo_fixup: Leaving currVFO alone
rig.c(1846) vfo=currVFO, curr_vfo=None
rig_get_freq: cache miss age=1000000ms, cached_vfo=None, asked_vfo=None
rig_get_freq(1928): vfo_opt=0, model=3072
icom_get_freq called for None, curr_vfo=None
icom_get_freq: using vfo=None
3:frame.c(425):icom_transaction entered
icom_transaction: cmd=0x25, subcmd=0x00, payload_len=0
4:frame.c(138):icom_one_transaction entered
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1
write_block(): TX 7 bytes, method=2
0000 fe fe 90 e0 25 00 fd ....%..
read_string_generic called, rxmax=200 direct=1
read_string_generic: minlen now 7
read_string_generic(): RX 7 characters, direct=1
0000 fe fe 90 e0 25 00 fd ....%..
icom_one_transaction: DEBUG retval=7, frm_len=7, cmd=0x25
read_string_generic called, rxmax=200 direct=1
read_string_generic(): RX 6 characters, direct=1
0000 fe fe e0 90 fa fd ......
4:frame.c(352):icom_one_transaction returning(-9) rigctld.c(626) Startup: rigctld -vvvvv -m 3072 -r COM1
rigctld Hamlib 4.5~git Tue May 17 22:03:48 2022 +0000 SHA=468dd1
Report bugs to [hamlib-developer@lists.sourceforge.net](mailto:hamlib-developer@lists.sourceforge.net)

n4ogw, 2024-06-24:
What version of gpredict? With recent versions I had to enable vfo mode in hamlib (add -o to rigctld command line).

curtiswilliams, 2024-06-27:
So -o instead of —vfo?Curtis Williams

mdblack98, 2024-06-28:
Yes...short form and long form....long form may end up being the only solution once we run out of the useable characters.
Mike W9MDB

rigctl -h
Usage: rigctl [OPTION]... [COMMAND]...
Send COMMANDs to a connected radio transceiver or receiver.

  -m, --model=ID                select radio model number. See model list
  -r, --rig-file=DEVICE         set device of the radio to operate on
  -p, --ptt-file=DEVICE         set device of the PTT device to operate on
  -d, --dcd-file=DEVICE         set device of the DCD device to operate on
  -P, --ptt-type=TYPE           set type of the PTT device to operate on
  -D, --dcd-type=TYPE           set type of the DCD device to operate on
  -s, --serial-speed=BAUD       set serial speed of the serial port
  -c, --civaddr=ID              set CI-V address, decimal (for Icom rigs only)
  -t, --send-cmd-term=CHAR      set send_cmd command termination char
  -C, --set-conf=PARM=VAL       set config parameters
  -L, --show-conf               list all config parameters
  -l, --list                    list all model numbers and exit
  -u, --dump-caps               dump capabilities and exit
  -o, --vfo                     do not default to VFO_CURR, require extra vfo arg
  -n, --no-restore-ai           do not restore auto information mode on rig
  -i, --read-history            read prior interactive session history
  -I, --save-history            save current interactive session history
  -v, --verbose                 set verbose mode, cumulative (-v to -vvvvv)
  -Y, --ignore-err              ignore rig_open errors
  -Z, --debug-time-stamps       enable time stamps for debug messages
  -h, --help                    display this help and exit
  -V, --version                 output version information and exit
  -!, --cookie                  use cookie control

Either one. -o and --vfo do the same thing.