csete/gpredict #365: Rotator Control: Polar view orientation incorrect?

cdavidson0522, 2024-08-28:
The polar view shown in the rotator control window is displayed different than the polar view in the main window. 

[image]

My selected polar orientation is S/W/N/E.

[image]

csete, 2024-09-17:
Looks like a bug, thanks for reporting.

cdavidson0522, 2024-09-27:
I noticed the polar view in the next pass details exhibits the same display issue.

csete, 2026-08-16:
Unfortunately, there is no easy way to fix this because the polar plots in the pass dialog and rotator controller use the global settings. As a temporary workaround, you can change the setting in the global preferences (Edit➔Preferences➔Modules➔Polar View).