csete/gpredict #367: ICOM IC-910H doppler control semi-functional with current hamlib

dl3jop, 2024-09-18:
Hey,

I've seen multiple issues, some closed, some still open, with conflicting issues regarding doppler control on the IC-910.
I did some tests and wanted to share the current state (and if i find ideas how to fix it) :

Issue:

- Doppler control with gpredict/hamlib/ic910 not working
- only downlink is corrected, rigctld shows rejected command errors
- if the VFO knob on the radio is turned, the frequency jumps back and forth a couple of kHz
- disengages after some seconds

Steps to reproduce:

- `rigctld Hamlib 4.6~git 2024-08-19T14:01:05Z SHA=137231 64-bit`
- gpredict 2.2.1
- radio setup as per https://github.com/csete/gpredict/blob/master/doc/notes/ic-910.txt:
[image]
- rigctld commands:
  - rigctld -m 3044 --serial-speed=19200 -r /dev/ttyUSB0 -vvvvv
  - adding --vfo has no apparent effect
  - adding --twiddle_timeout=2s decreases the frequency jumps a little but sometimes jumps still occur
- IC-910 in satellite mode
- Result:
  - gpredict sets inital downlink frequency, and tracks it for some seconds, than disengages

The rigctld log is spammed with rejected commands: https://pastebin.com/asNU0FjC
The logs show errors when addressing VFO_A/B but in this mode I think it should be Main/Sub?

I also tried this [branch](https://github.com/KaComet/gpredict/tree/Hamlib_VFO) with no success

Did anyone get it to work lately?
All the best,
Joshua

mdblack98, 2024-09-18:
Hopefully you are building you own Hamlib.
Replace the ic910.c file with the attached and see if it behaves better.

 
 

   

   
grafik.png (view on web)

  
 *  Hamlib CI-V backend - description of IC-910 (VHF/UHF All-Mode Transceiver)
 *  Contributed by Francois Retief ***@***.***>
 *  Copyright (c) 2000-2010 by Stephane Fillod
 *
 *
 *   This library is free software; you can redistribute it and/or
 *   modify it under the terms of the GNU Lesser General Public
 *   License as published by the Free Software Foundation; either
 *   version 2.1 of the License, or (at your option) any later version.
 *
 *   This library is distributed in the hope that it will be useful,
 *   but WITHOUT ANY WARRANTY; without even the implied warranty of
 *   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 *   Lesser General Public License for more details.
 *
 *   You should have received a copy of the GNU Lesser General Public
 *   License along with this library; if not, write to the Free Software
 *   Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA  02110-1301  USA
 *
 */

#include <stdlib.h>

#include <hamlib/rig.h>
#include "icom.h"
#include "icom_defs.h"
#include "frame.h"
#include "idx_builtin.h"
#include "tones.h"

/*
 * It seems some IC910 out there have weird firmware. Uncomment the following
 * if your modes are wrong, and please report to hamlib-developer maillist
 * with firmware number. That'd be interesting to have a word from Icom
 * on this subject, and if firmware updates are possible.
 */
#ifdef HAVE_WEIRD_IC910_MODES
static int ic910_set_mode(RIG *rig, vfo_t vfo, rmode_t mode, pbwidth_t width)
{
    /* FIX: The IC-910 has "Set FM" = 4, which is RTTY in for other radios */
    if (mode == RIG_MODE_FM)
    {
        mode = RIG_MODE_RTTY;
    }

    return icom_set_mode(rig, vfo, mode, width);
}

static int ic910_get_mode(RIG *rig, vfo_t vfo, rmode_t *mode, pbwidth_t *width)
{
    /* FIX: The IC-910 has "Set FM" = 4, which is RTTY in for other radios */
    int retval = icom_get_mode(rig, vfo, mode, width);

    if (*mode == RIG_MODE_RTTY)
    {
        *mode = RIG_MODE_FM;
    }

[... 274 lines cut ...]

    .set_mem =  icom_set_mem,
    .vfo_op =  icom_vfo_op,
    .scan =  icom_scan,
    .get_dcd =  icom_get_dcd,
    .decode_event =  icom_decode_event,
    .rig_open =  icom_rig_open,
    .rig_close =  icom_rig_close,
    .set_split_vfo = icom_set_split_vfo,
    .set_split_freq = icom_set_split_freq,
    .get_split_freq = icom_get_split_freq,
    .set_split_mode = icom_set_split_mode,
    .get_split_mode = icom_get_split_mode,
    .set_split_freq_mode =  icom_set_split_freq_mode,
    .get_split_freq_mode =  icom_get_split_freq_mode,
    .set_ctcss_tone =  icom_set_ctcss_tone,
    .get_ctcss_tone =  icom_get_ctcss_tone,
    .set_ctcss_sql =  icom_set_ctcss_sql,
    .get_ctcss_sql =  icom_get_ctcss_sql,
    .set_rptr_shift =  icom_set_rptr_shift,
    .set_rptr_offs =  icom_set_rptr_offs,
    .get_rptr_offs =  icom_get_rptr_offs,

    .hamlib_check_rig_caps = HAMLIB_CHECK_RIG_CAPS
};

dl3jop, 2024-09-18:
Hey Michael, 
thanks for coming back so quickly, I compiled it with your version with the new backend but the error still persists:

Using --vfo the rigctl gets unresponsive trying to retreive the VFO:
`Backend version: 20240831.2, Status: Stable
main: Using IPV4
main: rigctld listening on port 4532
Connection opened from 127.0.0.1:60392
mutex_rigctld: client lock engaged
mutex_rigctld: client lock disengaged
mutex_rigctld: client lock engaged
mutex_rigctld: client lock disengaged
handle_socket: doing rigctl_parse vfo_mode=1, secure=0
rigctl_parse: called, interactive=1
rigctl_parse: cmd=S(53) handle=9
rig_parse_vfo called
`

Discarding the --vfo flag yields the ` Command rejected by the rig` errors as in the logs above

mdblack98, 2024-09-18:
--vfo is the only way to run Icom rigs with gpredict -- without the vfos can swap like crazy.
Can you send me a debug log with the --vfo please?

Message ID: ***@***.***>

dl3jop, 2024-09-18:
Ah, thanks for the informationabout the --vfo.

Here's the log: https://pastebin.com/DvC3bS6p

After engaging control in gpredict it just freezes at `rig_parse_vfo called
`

mdblack98, 2024-09-18:
> Ah, thanks for the information about the --vfo flag.

Are you using this branch?
https://github.com/KaComet/gpredict/tree/Hamlib_VFO

mdblack98, 2024-09-18:
Or csete's repo?
https://github.com/KaComet/gpredict.git

Most recent commit:
commit 91a4a3fb15e7eab0374d1bb7c859d386818b48eeAuthor: AsciiWolf ***@***.***>Date:   Tue Jul 9 16:06:28 2024 +0200
    Desktop fixes

 
 

https://github.com/KaComet/gpredict/tree/Hamlib_VFO

dl3jop, 2024-09-18:
Using the master branch it freezes on the same position: https://pastebin.com/dxfw3zPZ

Using the hamlib_VFO branch, it tracks  and corrects downlink but the uplink is unchanged and not written to the radio. When spinning the radios VFO dial it crashes after some seconds:
[vfo_branch.log](https://github.com/user-attachments/files/17050089/vfo_branch.log)

mdblack98, 2024-09-22:
I assume you are compiling yourself?

Can you try the latest git version of hamlib?

Mike W9MDB